mjg59/python-broadlink #110: how to provision SP2 device for my wifi?

Cyber83, 2017-06-21:
Hi,

I am trying to get my Broadlink SP2 to my network.
I do not have an android or iphone, so I hoped I can setup the smart plug using this lib.

I am able to connect with the BroadlinkProv Hotspot, and there I can control the power with
import broadlink
mac="XXXXXXXX".decode('hex')
devices = broadlink.sp2(('192.168.10.1', 80),mac)
devices.auth()
devices.set_power(True|False)
print devices.check_power()

but the example usage for setup does nothing:
import broadlink
broadlink.setup('myssid', 'mynetworkpass', 3)

is there anything else I need to do?
my wifi SSID is hidden, but I guess this should not be an issue?

thanks a lot!

BR
Alois

Cyber83, 2017-06-22:
Hi,

I have borrowed an android phone to configure the smartplug, it is now in my network and is well controllable with python-broadlink.

Thanks for your work!

BR
Alois

frostworx, 2017-10-28:
hi, 
the example just worked fine, after I disabled my main network connection on the setup machine temporarily (so the connection to the broadlink was the only and therefore the default one).
Apparently too late for you but maybe it helps someone else :}

large, 2018-09-04:
On my Broadlink RM2 Pro it does not connect to a hidden SSID.
I had to put it into something visible...

closed 2024-04-21 by felipediel