mjg59/python-broadlink #113: Can't Enter Learning Mode

mrjemson, 2017-07-02:
Attempting to enter learning mode throws an error as below;

Python 2.7.12 (default, Nov 19 2016, 06:48:10)
>>> import broadlink
>>> devices = broadlink.discover(timeout=5)
>>> devices[0].auth()
True
>>> devices[0].host
('192.168.1.155', 80)
>>> devices[0].enter_learning()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: device instance has no attribute 'enter_learning'

mjg59, 2017-11-25:
Your device isn't being detected as an RM. What does the broadlink_discovery script say?

mrjemson, 2017-11-26:
Thanks for coming back to me. Please see below:

~/python-broadlink-master/cli# ./broadlink_discovery
discover
###########################################
Unknown
# broadlink_cli --type 0x2712 --host 192.168.1.155 --mac 94fdb00d43b4
Device file data (to be used with --device @filename in broadlink_cli) :
0x2712 192.168.1.155 94fdb00d43b4
~/python-broadlink-master/cli#

mjg59, 2017-11-26:
Hm, ok. Can you edit broadlink/__init__.py and add

print(devtype)

at the beginning of the gendevice() function?

mrjemson, 2017-11-26:
20010 is returned with that added.

mjg59, 2017-11-27:
Ok, in the gendevice() function try adding
```if devtype == 0x4e2a:
    return sp2(host=host, mac=mac)```
and see if that works?

mrjemson, 2017-11-29:
Getting the same error, except the last line states "sp2 intance" instead of "device instance"
`AttributeError: sp2 instance has no attribute 'enter_learning'`

mrjemson, 2017-11-29:
I just tried this and it got passed that line;
`if devtype == 0x4e2a:
    return rm(host=host, mac=mac)`

Temp always returning 1.0 and check_data() always returning the same no matter what I press in learning_mode.

```
>>> devices[0].check_temperature()
1.0
>>> devices[0].check_data()
'\x01\x00\x00\x00\x00\x00C\xff\x00\x00\x00\x00'

```

mjg59, 2017-11-30:
Ah, sorry, you're right - that should have been rm, not sp2. I'm afraid I don't have an idea why it doesn't work otherwise, though. What was this device sold as?

muriloyvens, 2018-08-26:
hello, any news about thoe RM pro+ problem?
still recognizing as 'Unknown' with error

stachdude, 2019-06-16:
Hi,

sorry to dig up the old potato, but turns out that the AUX AC seems to be using this exact DeviceID as its controller.

Is there any chance that we could work on this a bit ?

ThisIsQasim, 2020-04-25:
Same issue. Kenwood AC with built in Broadlink device of devtype 20010 currently discoverable but nothing else.

liaan, 2020-06-01:
> Same issue. Kenwood AC with built in Broadlink device of devtype 20010 currently discoverable but nothing else.

Possible you can test https://github.com/liaan/broadlink_ac_mqtt, would like to see if works with kenwood ac’s as well

ThisIsQasim, 2020-06-01:
@liaan I have never used an MQTT server before, any pointers on how I can test?

liaan, 2020-06-01:
> @liaan I have never used an MQTT server before, any pointers on how I can test?

Ill assist tomorrow, ill update code so dont need mqtt for testing.  
If reads the AC correctly one can try make it work

mrjemson, 2020-06-02:
> 
> Possible you can test https://github.com/liaan/broadlink_ac_mqtt, would like to see if works with kenwood ac’s as well

Hi Liaan, I am testing this now and have an MQTT server. I can see in config.yml there is a device section where I assume you can add the IP of your unit in there? Can you please advise on formatting here? My AC is on a different subnet so won't be organically discoverable, if needed I can temporarily connect something to the network the AC is on to test though.

mrjemson, 2020-06-02:
@liaan I ended up setting up a client on the same network as the AC as I was excited to test this.
Looks to detect the AC, however doesn't look like the MQTT component will auth despite the username and password fields in the config file. Will open up an issue on your repo to delve into this a bit further.

liaan, 2020-06-02:
> Possible you can test https://github.com/liaan/broadlink_ac_mqtt, would like to see if works with kenwood ac’s as well
>
> Hi @liaan, I am testing this now and have an MQTT server. I can see in [...]

The config file don't support device yet.   Will work on it this week.   I also answered the other question on the issue you logged. ( https://github.com/liaan/broadlink_ac_mqtt/issues/5)

ThisIsQasim, 2020-06-02:
Hi @liaan, So I set up a Mosquitto mqtt broker and your plugin seems to be reporting the temperature and all other values properly. I can also control the AC by publishing to mqtt topics. I am however having a difficult time getting it to work with the homebridge-mqttthing plugin but that is probably unrelated.

ThisIsQasim, 2020-06-02:
homebridge is used for HomeKit so the ha platform would be HomeKit. Anyways, it turns out a lot of the things are a hit or miss including fanspeed which doesn’t update instantly but after a minute or so and fixation doesn’t report correctly and sometimes the ambient temperature just reports as 0

liaan, 2020-06-02:
> homebridge is used for HomeKit so the ha platform would be HomeKit. Anyways, it turns out a lot of the things are a hit or miss including fanspeed which doesn’t update instantly but after a minute or so and fixation doesn’t report correctly and sometimes the ambient temperature just reports as 0

Ok, cool.   

The issue with updates is that the AC cannot deal with multiple requests at same time (phone app and script) and simply hangs for a bit.   Ive tweaked the code as much possibly since yesterday to make it more reliable.  Currently it will only update every 30 seconds if changes are made using other device(remote/phone) as logic is that you not going to really change settings really often so dont need be 100 in sync. This also to reduce requests to AC

I run openhab with homekit Plugin and latest code been 100 stable since yesterday..  

Not sure how to help otherwise..   maybe homebridge guys can use the AC protocol definition to make native plugin, its all MIT licensed, just need volunteers :)

liaan, 2020-06-02:
> homebridge is used for HomeKit so the ha platform would be HomeKit. Anyways, it turns out a lot of the things are a hit or miss including fanspeed which doesn’t update instantly but after a minute or so and fixation doesn’t report correctly and sometimes the ambient temperature just reports as 0

Did you try https://github.com/makleso6/homebridge-broadlink-heater-cooler yet? its based on my code from my AC so it might work on your AC

ThisIsQasim, 2020-06-03:
@liaan I am trying the new code. I can live with the low refresh rate but it is still not reporting fixation properly, always returns 0 no matter what. Also aware of the homebridge plugin but that is just as buggy.

liaan, 2020-06-03:
> @liaan I am trying the new code. I can live with the low refresh rate but it is still not reporting fixation properly, always returns 0 no matter what. It is also currently reporting ambient temperature as zero.
> Also aware of the homebridge plugin but that is just as buggy.

I just a big update ... some of the topics has changed (mode/fanspeed), so just keep eye on it.  Also added config for devices so you should be able to be on different broadcast domain

if ambient is still wrong .. you can try run it with the -d option, it will do full debugging of all steps, maybe there something not responding/timeing out or wrong format

ThisIsQasim, 2020-06-03:
Yes the -d option is dumping some hex and still reporting ambient temperature as 1 & fixation_v as 0

paradoxm, 2020-11-29:
I bought RM4 pro and it has `dev_type = 0x653c`. I set `0x653C: (rm4, "RM4 pro", "Broadlink"),` into the file `brodlink/__init__.py` and now it works as well

felipediel, 2021-02-17:
Fixed somewhere in the past. Closing.

closed 2021-02-17 by felipediel