mjg59/python-broadlink #114: Is any way to have the energy monitoring of the sp2？

xyyaya, 2017-07-03:
Is there any idea or method to get the energy monitoring infomation of the sp2?

VooDooN1, 2017-07-10:
I need the same functionality for SP3S. I don't know how to extract bytes with power information from payloads.

eschava, 2017-07-27:
I've implemented energy monitoring for SP3S in my clone of the python-broadlink (and pulled request of course) but there is a doubt that it will work for SP2 in the same way

VooDooN1, 2017-07-28:
**xyyaya**, can you test the code from **eschava** for SP2? I have SP3S and this code works fine, but SP2 of my friend (0x2711) does not receive the energy value. Thank you! We use the same algorithm for PHP-Broadlink library in the MajorDoMo project.

closed 2017-11-25 by mjg59