mjg59/python-broadlink #118: discover returning empty list

codelayout, 2017-07-15:
Python 2.7.6 (default, Oct 26 2016, 20:46:32) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> broadlink.discover(timeout=5)
[]

I am able to talk to the device (RM3 pro) using RM Bridge, and I can ping it from the laptop (both on the same network using WPA2/AES)
any suggestions?

onepas, 2017-09-03:
Like me.
Official eControl app still get list of devices, 
our device is RM pro

mob41, 2017-09-09:
Please spend some time reading the following issue and see whether it helps.

https://github.com/mjg59/python-broadlink/issues/93

thias, 2018-01-25:
I have the same issue with a brand new "RM Pro" device and python-broadlink 0.6. I have looked at #93 and #82, which have similar symptoms, but seems to have been fixed already.
```
$ python3
Python 3.6.3 (default, Oct  9 2017, 12:07:10) 
[GCC 7.2.1 20170915 (Red Hat 7.2.1-2)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> devices = broadlink.discover(timeout=30)
>>> print(devices)
[]
>>> 
```
Everything before that worked fine, I was able to connect it to my Wi-Fi network using `broadlink.setup()`, and I can see some "communication" going on when running `broadlink.discover()` : 
```
# tcpdump -nn -i any -x -p udp port 80
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
10:23:48.982303 IP 192.168.19.6.48325 > 255.255.255.255.80: UDP, length 48
	0x0000:  4500 004c 44f7 4000 4011 21fc c0a8 1306
	0x0010:  ffff ffff bcc5 0050 0038 b774 0000 0000
	0x0020:  0000 0000 0100 0000 e207 170a 1204 1901
	0x0030:  0000 0000 c0a8 1306 c5bc 0000 f2c2 0000
	0x0040:  0000 0600 0000 0000 0000 0000
10:23:49.084749 IP 192.168.19.128.80 > 192.168.19.6.48325: UDP, length 128
	0x0000:  4500 009c 076a 0000 ff11 0c10 c0a8 1380
	0x0010:  c0a8 1306 0050 bcc5 0088 a67d 0000 0000
	0x0020:  0000 0000 0100 0000 e207 170a 1204 1901
	0x0030:  0000 0000 c0a8 1306 c5bc 0000 16d4 0000
	0x0040:  0000 0700 0000 0000 0000 0000 5ecf 9f66
	0x0050:  8727 8013 a8c0 e4d8 fb0d 43b4 e699 bae8
	0x0060:  83bd e981 a5e6 8ea7 0000 0000 0000 0000
	0x0070:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0080:  0000 0000 0000 0000 0000 0000 0000 0000
	0x0090:  0000 0000 0000 0000 0000 0200
^C
2 packets captured
2 packets received by filter
0 packets dropped by kernel
```
Before resetting the device, it had worked fine with the official Android App and its cloud-based communication, so the device itself works (but that's not how I want to use it, I want local-only).

thias, 2018-01-25:
I continued digging a bit, but still couldn't get discovery to work. I had a look at #95 and tested that change (the 3rd one seems already merged, not the 2 first), then I also tested with python 2.7 with pycrypto and pyaes, but none worked. I could still see the UDP port 80 exchange, but the device list was empty....

...then it struck me : This is UDP, and with 255.255.255.255 involved! Not something standard and surely problematic as far as connection tracking goes. So I just added this temporary firewall rule, and it started working (I'm running Fedora 27) : 
```
iptables -I INPUT 1 -p udp --sport 80 -j ACCEPT
```
Note that I did also need the changes from #95 to get things working with python 3.6, but now it's all good :
```
$ ./broadlink_discovery 
discover
###########################################
RM2
# broadlink_cli --type 0x2712 --host 192.168.19.128 --mac e4d8fb0d43b4
Device file data (to be used with --device @filename in broadlink_cli) : 
0x2712 192.168.19.128 e4d8fb0d43b4
temperature = 22.0
```
Woohoo! 😄

felipediel, 2021-02-17:
Hello. Two possibilities:
1. Your devices are locked. When the devices are locked they cannot be discovered. To unlock:
```python3
import broadlink as blk
>>> d = blk.hello('192.168.0.16')  # Example device
>>> d.auth()
>>> d.set_lock(False)
```
2. You have multiple network interfaces and didn't specify an appropriate local_ip_address or discover_ip_address, so the OS is binding the socket to the wrong interface. The best way to discover is using the L3 broadcast address:
```python3
>>> import broadlink as blk
>>> devs = blk.discover(discover_ip_address='192.168.0.255')
>>> devs
```
I am closing old issues. If it doesn't work, you can reopen it.

closed 2021-02-17 by felipediel