mjg59/python-broadlink #121: SPmini T1

Mario1234, 2017-08-10:
I think this API doesn't support model SPmini T1.
I recorded some raw data about what e-control Android app sends to power on the smart plug and what it receives back.
e-control sends this:
f1 00 01 76 00 00 00 04 11 02 59 ca 9c dd 93 2e 
36 fc 39 fd be 01 dd a2 e2 34 92 ad b7 c1 a9 a4 
af e4 d9 37 38 41 93 e2 ec 89 68 35 4e c1 75 5e 
19 c6 6c 9c 77 f2 ef 5a a6 fd c3 ef 4e 1a b5 7b 
9b 47 9f 21 fd 48 f2 c4 b6 45 dd 51 fc 06 3b cd 
cd bb e4 25 c7 46 a0 3e ee 80 ac eb 5c 4c e1 e0 
d3 3e e7 9e f0 86 b0 61 a0 fd 0b b6 2a e8 65 dd 
5d 46 42 a7 d5 86 1e af 41 fc 45 93 ec ab 98 3c 
8a e4 3e de ff 7e b9 0b 66 dd 62 02 a7 bf e7 08 
4c d9 c9 74 29 ad 1a b5 55 c6 47 36 e8 68 39 9d 
18 96 4a ba e8 2f b4 c4 90 b8 05 0e 62 2e f7 23 
65 07 91 81 68 09 91 5a d5 ec 57 ec cf f4 52 f6 
ed e1 b4 57 dd ec 00 4f db 64 16 f4 7f 1f 5e ad 
4c bc fc 8e ed 1d 7d 1e fe 80 a6 16 44 6f e5 68 
7b 8a 36 80 9f 6b 97 de 64 ad 1f 3b 2d c2 54 0e 
99 6b 78 3a 1a c8 a0 1f 1e 8d 2c 7f c6 0d f3 86 
86 b2 03 3b 3c 18 df 96 1c a4 d1 0a f3 e1 aa 7e 
db 8a 18 b8 05 72 d7 70 f7 18 f0 f1 e3 f7 26 38 
97 a1 62 82 c9 c3 fd 90 b4 dc c3 03 44 9f d7 f5 
35 02 d8 46 87 e9 c2 cc ff 87 45 a5 57 18 97 54 
c6 e2 65 28 62 79 51 ed 74 1a d4 fc 55 f4 d2 61 
b0 fc 54 46 a5 57 a2 ea b3 4f 77 33 c3 40 f5 41 
bb f7 8c 9d 48 48 79 8f 81 6c bf 48 15 16 49 e3 
71 ee 03 a0 02 16 b2 1c d8 72

And the plug answer this:
f3 00 00 aa 00 00 38 40 00 74 00 00 38 40 00 00 
ef 00 fe 00 f1 00 fc 00 ac 00 20 00 9a 00 3a 82 
90 8d 6a 9b bd 5c 78 69 fc b2 5a 87 59 4b 23 79 
39 85 98 8f 5f 79 8e fc 47 24 af db 64 fd ff 31 
e6 06 02 93 e5 fb 39 a9 85 ee 57 da 06 d5 6e e2 
c5 63 7d 18 b1 5e ae 70 d7 43 0f f3 e7 33 b8 47 
d0 d7 02 fd 03 d2 de 89 85 3e ba b1 28 0a b7 b1 
70 d3 b0 8f 8c 3c e0 81 2b 98 80 58 d5 93 07 4d 
93 bf 7e 7a 97 f3 6e e6 d1 a8 b5 91 5a f4 ee 59 
47 b2 94 fd 0f 91 1b 3a de 20 8b 09 03 15 b4 1e 
0c 88 b4 f6 f8 0d 8c 74 c1 f5 03 44 ca 89 f4 00 
00 c4 c8 53 99 1b 39 36 11 60 56 c5 a1 40 69 16 
1e 06 d6 fe 01 43 8a e4 44 46 b2 3c ee 63 6d 07 
78 1c cb 7f b4 78 b4 0a 72 3e 12 67 37 e5 ff d7 
76 e6 3d fa b2 1b c2 d5 72 ed 00 55 a3 44 f2 e3 
24 6d 2e 72 dc 43 aa 98 77 0f d9 c1 88 29 15 e1 
22 de 49 d1 d9 7e 5b 63 ea 97 f6 46 ea 4c d8 43 
24 4e 2c 6d 53 5a 7f 66 9a 6c 9e 51 f5 ca de a4 
55 14 76 a5 26 a9 66 3b f3 ae 98 6b 6e 57 77 dc 
46 ba 7b cd 32 3d 05 a4 8d e3 6f 7d e5 49 de 24 
6e b1 ff ba f7 f4 0b e1 61 84 fa f8 23 1e f2 85 
81 f1 ff 09 c0 aa 38 2b 3a ab a1 6c 42 65 d9 f9 
dc 8a 4a 49 93 9e 49 3b 6f 60 f7 e5 f4 a5 6f 4e 
64 09 1d ec 77 76

As additional info I recorded it at day 10 of month 8 year 2017 11:40:09
I hope anyone who wanna do reverse engineering might find this interesting.
The only thing I could realized is that when you plug the device it starts to send
a hello message of 48 bytes saying its MAC address, not encrypted by port 15001 udp.
So you can get MAC address from position 42 to 47 like:
macAddress[0]=receivedData[47];
macAddress[1]=receivedData[46];
macAddress[2]=receivedData[45];
macAddress[3]=receivedData[44];
macAddress[4]=receivedData[43];
macAddress[5]=receivedData[42];

Mario1234, 2017-08-18:
I made some coding in Java and C++ that works in the same way as this broadlink API.

SPmini T1 worked with:
[https://github.com/mjg59/python-broadlink](url)
[https://github.com/Mario1234/java-broadlink](url)
[https://github.com/Mario1234/cpp-broadlink](url)

Maybe someone find my codes usefull

HyperDevil, 2018-04-23:
Hi,
I bought a clas ohlson home - wifi smart plug
According to the DHCP information: Broadlink_Outlet-T1-59-1a-a3
So i assume its a T1 just like yours.

UDP port 16680:
00:00:00:00:00:00:02:00:00:00:e2:07:25:30:0e:01:17:04:00:00:00:00:15:03:00:0a:8e:8a:00:00:59:c1:00:00:00:00:06:00:00:00:00:00:00:00:00:00 

17:04 is probably the time

Anybody got this working in python without using the clas ohlson home app?

felipediel, 2021-03-02:
Hi @Mario1234. Do we have this device [here](https://github.com/mjg59/python-broadlink/blob/b33bbdbbcf1270fb77f346f0e003273247d3ff2f/broadlink/__init__.py#L17-L111)? If not, could you please tell me the devtype (PID) so I can add it?