mjg59/python-broadlink #131: Learn not work properly in some devices

s3Pol, 2017-12-03:
With the IR command Ericsson Mediaroom the learn mode don't work properly.
For instance if I send the code that I just had learned the set-top box don't take any action.

Example:
./broadlink_cli --type 0x2712 --host 192.168.1.151 --mac 3bff4234ea34 --learn
Learning...
260034000e111712171218060e1218110d0719110d0718070e120d000b610e111712181217060e1218120d0619120d0619060e120d000d0500000000
./broadlink_cli --type 0x2712 --host 192.168.1.151 --mac 3bff4234ea34 --send 260034000e111712171218060e1218110d0719110d0718070e120d000b610e111712181217060e1218120d0619120d0619060e120d000d0500000000

felipediel, 2021-02-18:
If you use the official app to learn the same command, does it work?

felipediel, 2021-03-12:
I will consider this case closed, if you still have the issue please reopen it.

closed 2021-03-12 by felipediel