mjg59/python-broadlink #133: RM PRO+  cant get temperature

dimagoltsman, 2017-12-22:
i have a probably new device, model: RM PRO+
and getting temperature always show 0.0

khanwhlee, 2018-01-08:
I have the same prob as well...

dimagoltsman, 2018-01-08:
can be closed... i have a device that says "Rm pro +", the thing with the plus is that it doesnt have temp sensor

antonraharja, 2018-01-16:
Hi @dimagoltsman 

Thanks for the info, was looking for this. Can you share where did you get the info, or maybe you've tested it ? Thanks.

dimagoltsman, 2018-01-16:
@antonraharja 

i talked to Broadlink directly.

their answer:

> Dear Customer,
Pls kindly note the device of RM pro+ don't have the temp sensor function, so it can't help show the temp on it, thanks!

Dustyfrogz, 2018-04-10:
you are lucky, I can't use the rm pro + with jeedom or domoticz

Fusseldieb, 2018-12-16:
Why did they take this awesome function away? Bummer.

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel