mjg59/python-broadlink #135: Support SP2Mini2

Snake-Sanders, 2018-01-01:
Hi guys, 
Thank you for your work. I forked this project to work in the implementation for devices SP2Mini2. I do this just because the e-control app for android crashes every time I start it and it takes 20 to 30 seconds to start and makes me really nervous to wait that much to switch on the lights. 

Here is the "smart" power outlet picture:
[image]
I manage to send commands and receive responses but it does not work, I dont know where to put the set On/Off parameters.
After sniffing with Wireshark the QUIC packages I noticed that the packages do not follow any pattern on the length.

Here is what I am able to send and receive.

Index:   0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0000 =  5a a5 aa 55 5a a5 aa 55 00 00 00 00 00 00 00 00
0010 =  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020 =  9d e0 00 00 28 27 6a 00 b3 10 82 bb f4 34 ea 34
0030 =  00 00 00 00 bb c4 00 00 76 0a 14 e4 0c 75 49 12
0040 =  67 48 22 e1 a8 b0 bb 01 e8 de c9 98 e7 66 5d 63
0050 =  b1 7f 77 26 71 73 a1 92 86 8c 6d 72 b8 fc 42 42
0060 =  42 71 57 2c 9b de c3 a7

Dump buffer received...
Index:   0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F
0000 =  5a a5 aa 55 5a a5 aa 55 00 00 00 00 00 00 00 00
0010 =  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020 =  a8 cb f9 ff 28 27 ee 03 b3 10 82 bb f4 34 ea 34
0030 =  00 00 00 00 bb c4 00 00

In wireshark I get other responses length and I dont know how to interpret the other bytes. Anyone has the same device?

ptd006, 2018-01-03:
I am not familiar with this device however it looks like your response has no payload and from what I know any error free response should have response[0x22] and response[0x23] both zero.  Thus I think there is a problem with your command packet.

I have a Broadlink-based heating controller (photo: http://peter.windridge.org.uk/playing-with-cheap-iot-devices/20171120_163450.jpg) and yes the packets received are of different lengths.

Snake-Sanders, 2018-01-03:
Thank you for your response, this information helps, I am going to check the request package.

Snake-Sanders, 2018-01-07:
I managed to switch On and Off the SP2Mini2 from my laptop, I am so happy! 

I have done several changes in the __send_packet()_ function due to the differences in the data location in the message buffer. I figured by reading the original frames using Wireshark.

Now, I will not push the changes to this branch because I did serious changes in the code. 
Basically during the analysis of the code I split all the devices in separated files and added my owns. 

Anyway feel free to go to the branch I made and bring those changes here. Don't hesitate to contact me for questions or helping with the merge.

Thanks a lot for this project!

closed 2018-01-07 by Snake-Sanders

ptd006, 2018-01-07:
Great!  Splitting the device classes into separate files is what I actually intended to do, so would welcome that change.

I had a quick look at your fork and see you import all device classes when discovering the device.  However, for me, one of the benefits of splitting up the devices is that I would only import the relevant one.

Snake-Sanders, 2018-01-07:
I must say I am not a Python developer, surely I have made mistakes in there.

ptd006, 2018-01-09:
Me neither :)  It's simply that my target platform is an old phone (running QPython on Cyanogenmod) so want to minimise load time and RAM usage.

felipediel, 2020-06-09:
Hi @Snake-Sanders. Thanks for the contribution!

I'm working on an update in which the brand and model will be displayed to the user. In the IHC app your device is registered as an URANT SP1-UK, which doesn't seem to match your model.

Do you confirm the brand? Does the check_power() method works for you?

felipediel, 2020-06-09:
@Snake-Sanders Your device is not an SP2. It is a [Kebidu KBT001929-EU](https://www.lazada.com.my/products/smart-wifi-plug-power-socket-remote-timer-switch-for-iphone-for-xiaomi-android-enchufe-interruptor-wifi-domotica-eu-i459404648.html), which has the same DNA.

Snake-Sanders, 2020-06-09:
Hi Felipe,
Now I have an iPhone and the app works fine, the problem was the android
app that is why I needed a quick hack.
I don’t have the pay script running but I can give you the info of the
devices.
As far as I remember the check status func did not work.

Cheers

felipediel, 2020-06-09:
@Snake-Sanders Do you mind helping me find out if your device is an SP1 or SP2-compatible device?

### How?
1. Download [debug1.py](https://drive.google.com/file/d/1on9g3Qbk6iPZswDmt2kVsI_omEDy-lge/view?usp=sharing) and [debug2.py](https://drive.google.com/file/d/1w3LWaPzEcl7ys684CDFD1uJINd4CPn6D/view?usp=sharing).
2. Open a terminal and type `python3 debug1.py` and `python3 debug2.py`.
3. Send me `debug1.txt` and `debug2.txt`.

These files will help me understand the DNA of your device, so we can choose the right class to control it with this library and Home Assistant.