mjg59/python-broadlink #136: Who is 智能遥控  aka Tomoyo Yukari

a-lurker, 2018-01-04:
The response to the tx'ed Broadlink discovery message and the response to the tx'ed temperature message for the 'RM2 Pro Plus2' (ie id '0x2787') both contain the hex sequence 'e699bae883bde981a5e68ea7'. This seems to be UTF8 for some Japanese characters (or possibly Chinese) that according to Google, translate as 'Tomoyo Yukari' . No idea if this makes any sense but Chinese and Japanese readers out there may have further info?

Snake-Sanders, 2018-01-04:
Well, since the application is developed in China, I assume it is Chinese

roiff, 2018-01-07:
@a-lurker is Chinese, 智能遥控 Means intelligent remote control

a-lurker, 2018-02-10:
@roiff 
Thanks for your answer - it's interesting how four hànzì characters represents three words. I have another one that starts with WiFi:      WIFI控制盒
Any translations for the above?

zhuwei-me, 2018-02-11:
@a-lurker WIFI控制盒 = WiFi Controlled Switch

closed 2018-03-18 by mjg59