mjg59/python-broadlink #140: Get_energy() not working with python3

bhaap, 2018-01-05:
Hi

Im trying to write an component to home assistant to get the power of SP2 and SP3s but the use of ord() in the get_energy() is inhibiting me from using this lib (home assistant is python3)

Any way to make it not use ord?

bhaap, 2018-01-05:
I remade the code into this to make it work:

```
  def get_energy(self):
    packet = bytearray([8, 0, 254, 1, 5, 1, 0, 0, 0, 45])
    response = self.send_packet(0x6a, packet)
    err = response[0x22] | (response[0x23] << 8)
    if err == 0:
      data = {}
      payload = self.decrypt(bytes(response[0x38:]))
      data['energy'] = int(hex(payload[7] * 256 + payload[6])[2:]) + int(hex(payload[5])[2:])/100.0
      return data
```

If someone wants they can use this (i dont know how) to implement a if statement with and without ord()

a-lurker, 2018-02-12:
Not being a Python programmer I'm not sure what's happening here (even after reading Python manual & examples). The [2:] syntax gets me.  It's element 2 to end - end of what?

Likewise what are the units kWh or Wh? What do bytes 7, 6 & 5 actually contain? Why the divide by 100.0 - would have thought it would be a divide by 10.0?   Also do these devices report power? Hoping I can be shown the light on this one. Then I might be able include it all in this code:

https://github.com/a-lurker/Vera-Plugin-BroadLink-Mk2

mjg59, 2018-03-18:
Fixed in 0.7

closed 2018-03-18 by mjg59