mjg59/python-broadlink #142: No SC1 support

SzymonZy, 2018-01-13:
Do you plan to add SC1 module  support to python plugin?

Snake-Sanders, 2018-01-16:
Hi
Is it SC1 or S1C? because apparently S1C is implemented.
Just for your information, each model may have slightly different protocol implementation, and the only way to find it out is having such model and test it. So either you provide us with the frames sniffed with Wireshark and we try to help you to implement it or you wait until somebody else with the same model comes here and wants to implement it.

SzymonZy, 2018-01-19:
I was thinking about this device https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F272837642698
Seller call it sc1

a-lurker, 2018-02-10:
The type id for the device linked above is 0x7547.

Out of interest;  the discovery reply message from a RM pro plus (id 0x2787) and a RM mini  (id 0x2737) contain the device's ip address little-endian at these locations just below the mac address:

ip = responsepacket[0x35:0x39]

However the SC1 (id 0x7547) store the ip address big-endian.

qinghuangchan, 2018-06-30:
Can confirm that SC1 behaves just like SP2.
`set_power` and `check_power` methods work just fine.

TomRom27, 2018-11-08:
Broadlnik Hub app's page shows that as SC1
http://4pda.ru/forum/lofiversion/index.php?t872550-20.html

howeve the app itself displays it as SP Mini

felipediel, 2020-06-20:
@a-lurker Thank you!

closed 2020-07-31 by Danielhiversen