mjg59/python-broadlink #152: Broadlink MP1 STATUS power switch does not work and Error getting temperature data from Broadlink device....Timeout

refe, 2018-03-03:
hello,
I am new to domoticz and to BroadLink power switches.
I just bought an MP1 and I want to control it with domoticz.
I manage to install broadlink module and add the Broadlink RM2 with Kodi Remote to domoticz, I have selected Device Type - Multiplug 1
the problem is that I can't control all the plugs through status switch , like when I press the button .
the individual switches  s1 to s4 are working fine

also I get  Error: (BroadLink) Error getting temperature data from Broadlink device....Timeout
I don't know why I get this error because this power switch does not have temperature sensor built in as far as I know.
also I don't have any device with temperature

this second error is frustrating...because I have set domoticz to send emails when I get errors..and now I have email send every 5 minutes.

refe, 2018-03-03:
if I create Dummy device and add 4 Virtual Sensor Switches with the start stop script like bellow

script:////home/domoticz/domoticz_server/scripts/python/bl_mp1_s4_start.py

#!/usr/bin/python
import broadlink
devices = broadlink.mp1(host=("192.168.1.25",80), mac=bytearray.fromhex("34ea34d0dcb1"))
devices.auth()
devices.set_power(4, True)

script:////home/domoticz/domoticz_server/scripts/python/bl_mp1_s4_stop.py

#!/usr/bin/python
import broadlink
devices = broadlink.mp1(host=("192.168.1.25",80), mac=bytearray.fromhex("34ea34d0dcb1"))
devices.auth()
devices.set_power(4, True)

I also create a general switch ...wich starts/stops all the S1-S4 ...but they are starte/stoped one after anoter ...not all at once.

script:///home/domoticz/domoticz_server/scripts/python/mp1_all_start.py
#!/usr/bin/python
import broadlink
devices = broadlink.mp1(host=("192.168.1.25",80), mac=bytearray.fromhex("34ea34d0dcb1"))
devices.auth()
devices.set_power(1, True)
devices.set_power(2, True)
devices.set_power(3, True)
devices.set_power(4, True)

found this start/stop scripts on http://domo-attitude.fr/broadlink-domoticz/

refe, 2018-03-03:
I also used the cli ...but no luck 

 ./broadlink_cli --type 0x2712 --host 192.168.1.25 --mac b1dcd034ea34 --turnoff
Traceback (most recent call last):
  File "./broadlink_cli", line 148, in <module>
    dev.set_power(False)
AttributeError: rm instance has no attribute 'set_power'

./broadlink_cli --type 0x2712 --host 192.168.1.25 --mac b1dcd034ea34 --turnon
Traceback (most recent call last):
  File "./broadlink_cli", line 142, in <module>
    dev.set_power(True)
AttributeError: rm instance has no attribute 'set_power'

Nightreaver, 2018-03-14:
The issue with using cli is that you might have used the `broadlink_discover` to find the cli commandline!?
Unless you use a fix like #157 , you have to type the proper device type on your own  
 `--type 0x2712` is **rmpro** device type, therefor has no `set_power`method

so you can try   
`./broadlink_cli --type 0x4EB5 --host 192.168.1.25 --mac b1dcd034ea34 --turnon`

if this doesnt work, you can also can also get the proper device type through the broadlink library

felipediel, 2021-02-17:
Closing, since the problem cannot be fixed in our code base.

closed 2021-02-17 by felipediel