mjg59/python-broadlink #154: [Question] - get_energy for SP3 smart plugs

wuid24, 2018-03-07:
Hi , 

first : its a wow !!! thanks for the developers that enabled this python based controlling - the broadlink app is miserable, and I cant count on the IFTT.  

my questions is for the support of getting the energy with SP3 - can anyone verify my thoughts that this option isn't hardware supported in this model ? 

thanks again,

eschava, 2018-03-07:
1. Please check sp2.get_energy() method. sp2 class is used for sp3 devices
2. use` '_'.join(format(s, '02x') for s in device.mac[::-1]` instruction  to show device.mac as human-readable string

wuid24, 2018-03-09:
thanks for #2

As for #1 i've tried it , and always get "0.01" so it seems that sp3 does not support energy readings..

Nightreaver, 2018-03-09:
energy is only supported for SP3S not for regular SP3
but instead SP3 has "night light" option that SP3S has not

wuid24, 2018-03-11:
ok thank you

closed 2018-03-11 by wuid24