mjg59/python-broadlink #165: Device discovery filter

horse315, 2018-03-24:
There's no way to filter device discovering by MAC, IP or type( Or any other API to create certain device instance by MAC or IP (but not both).

Nightreaver, 2018-03-28:
I can't get what you are trying to say what you want to achieve.

You can always discover all devices and then check if device ip/mac matches or you can create a totally new device by type as its done in the cli.

So maybe give a sample of what you wanna do

Nightreaver, 2018-03-28:
I am not sure if I understood correctly. So you want the discover to concentrate on a specific device mac, so save time. Because discovering every device might take longer ?

From my understanding, discovery doesn't work that way.
The broadlink "discover" will send a broadcast and receive and collect all the data it receives in response to that broadcast. There is nothing what will speedup the process - except reducing the timeout. Analyzing the data to get mac etc is already the last part.

The Thing that is taking time is the "broadlink_discovery" cli tool, because that will also talk to each device, authenticate etc.

You can skip that by just using the native broadlink.discover() , then you have direct access to the device list which does not take any longer the timeout you define. Then just pick the device you need. 
And you can lower the timeout if it works for you...

ptd006, 2018-10-30:
Agreed with Nightreaver.  
Another idea is to find the hostname of your device and obtain the IP address by simply looking up the hostname.  You can then .gendevice() your device manually.

If you insist on looking up with MAC address there are ways to do that.  However, I doubt they'd be much quicker than simply .discover()ing.

ptd006, 2018-10-30:
Here is an explicit example, which I have successfully tested with my broadlink central heating controller:

```
mac_addr = bytearray(b'\xxx\xxx\xxx\xxx') # you are supposed to know this ... 
hostname = "BroadLink_OEM-xx-xx-xx-xx" # and this

import socket
ip_addr = socket.gethostbyaddr(hostname)[2][0]

dev = broadlink.gendevice(0x4EAD, (ip_addr, 80), mac_addr)

dev.auth()
print('Current temperature: '+ str(dev.get_temp()) + 'C')

```

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/427. Now you can use broadlink.xdiscover() to create your own filter. Closing.

closed 2021-02-17 by felipediel