mjg59/python-broadlink #166: Discover does not work on rapsberry pi

sbsrouteur, 2018-03-24:
When using a Raspberry PI the discover function does not find the device.

I do not know if this is link to the raspberry, but the setup is as follows : 
-1 LAN NIC 
- 1 WIFI NIC (setup as access point)
the broadcom is setup on the PI wifi AP.

I add to change the code to bind to all IPs, and use the broadcast address of the wifi NIC. With these changes, the discover finds my device, and I can use the API services on the device.

Is there something I missed, or this configuration was not taken into account an requires a PR ?

Nightreaver, 2018-03-28:
The broadlink sadly is using an "automated" way to define the NIC that is used to discover the devices.
Take a loot at L#83 in __init__p.y file. You can manually change the source there and fill in the proper ip-adress you want to use for discover.

I'm not sure if it's possible to use 0.0.0.0 here, but feel free to test.
I had to change manually for my raspi setup as well.

sbsrouteur, 2018-03-31:
I have made 2 extra params to allow binding and broadcasting to the proper NIC. Would this justify a PR ?

Nightreaver, 2018-05-08:
you can always file a pr I guess :)

Best would be, if it doesnt affect anything when the param is not there and work well with default python python libraries

sbsrouteur, 2018-05-08:
Well I did file PR#173.

Unless I am mistaken, new parameters are optional, and do not prevent prior usage.

Nightreaver, 2018-05-08:
Ah I see... you're right

Does it make sense to have seperate "local" ip and broadcast ip?

sbsrouteur, 2018-05-08:
In my case it does because the packet is sent from the wired NIC IP when using the default broadcast IP, the answer comes back but somehow never comes back to the python script, unless the proper output address was used. (it might be computed from the localip, but this is over my python skills).

sbsrouteur, 2018-05-19:
Any hope to get this PR (#173) merged ?

eadmaster, 2019-08-22:
seems to be fixed now for me :)

felipediel, 2021-02-17:
Now we have a discover_ip_address option. Old issue. Closing.

closed 2021-02-17 by felipediel