mjg59/python-broadlink #171: check_nightlight with python3 not working

timmornYE, 2018-04-01:
In python2.7 all is working, but I need the lib in python3.5 and there I gett following error when calling `set_power`:

```
2018-04-01 20:34:57 ERROR (MainThread) [homeassistant.core] Error executing service <ServiceCall broadlink_switch.set_on>
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 1010, in _event_to_service_call
    await self._hass.async_add_job(execute_service)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/core.py", line 1007, in execute_service
    service_handler.func(service_call)
  File "/home/homeassistant/.homeassistant/custom_components/broadlink_switch.py", line 30, in set_on_service
    switch_all(True)
  File "/home/homeassistant/.homeassistant/custom_components/broadlink_switch.py", line 15, in switch_all
    devices[j].set_power(toOn)
  File "/srv/homeassistant/lib/python3.5/site-packages/broadlink/__init__.py", line 394, in set_power
  File "/srv/homeassistant/lib/python3.5/site-packages/broadlink/__init__.py", line 432, in check_nightlight
TypeError: ord() expected string of length 1, but int found
```

This seems to be a problem with code not adapted to python3. Look for example [here on stackoverflow](https://stackoverflow.com/a/41843740).

Nightreaver, 2018-04-02:
oh sorry... for that

I will look into it later today, hopefully come up with a proper solution

matisaul, 2018-05-08:
Hi @Nightreaver, could you look into this error? I'm using homeassistant and this issue cause me a lot of trouble. Thanks in advance!

Nightreaver, 2018-05-08:
it looks like it was fixed by @jonhen in #178 
please update to latest version and try again

Nightreaver, 2018-05-08:
So its actually not failing with `if type(payload[0x4]) == int:`

But its failing earlier at `if err == 0:` is that correct?

Can you post the full stacktrace again, as in your first post, please

matisaul, 2018-05-08:
I think is not entering in `if type(payload[0x4]) == int:` but in `if ord(payload[0x4]) == 1 or ord(payload[0x4]) == 3:`. It's very strange. This is the traceback:

```
Update for switch.living_luz_mesa fails
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity.py", line 204, in async_update_ha_state
    yield from self.async_device_update()
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/helpers/entity.py", line 327, in async_device_update
    yield from self.hass.async_add_job(self.update)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/switch/broadlink.py", line 293, in update
    self._update()
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/switch/broadlink.py", line 298, in _update
    state = self._device.check_power()
  File "/srv/homeassistant/lib/python3.5/site-packages/broadlink/__init__.py", line 418, in check_power
    if err == 0:
TypeError: ord() expected string of length 1, but int found
```

Nightreaver, 2018-05-08:
Can you add a line before 418, like that (2nd line)

```
    err = response[0x22] | (response[0x23] << 8)
    raise SystemExit(type(err))
    if err == 0:
```
and tell me what the type is ?
It will always "crash" when you add the line, so just add it, run it once, and remove afterwards!

I'm not sure if this is the issue... but it is weird that its crashing at comparison...

matisaul, 2018-05-08:
I put that line into the code, but it still showing the same error.
The output of the pip freeze is:
```
aiohttp==3.1.3
aiohttp-cors==0.7.0
argh==0.26.2
asn1crypto==0.24.0
astral==1.6
async-timeout==2.0.1
asyncio==3.4.3
attrs==17.4.0
beautifulsoup4==4.6.0
bitstring==3.1.5
boto3==1.6.1
botocore==1.9.1
broadlink==0.9
cachetools==2.0.1
certifi==2018.4.16
cffi==1.11.5
chardet==3.0.4
click==6.7
colorlog==3.1.2
cookies==2.2.1
cryptography==2.1.1
distro==1.2.0
docutils==0.14
ecdsa==0.13
enum-compat==0.0.2
envs==1.2.4
ephem==3.7.6.0
future==0.16.0
gTTS-token==1.1.1
home-assistant-frontend==20180426.0
homeassistant==0.68.1
http-ece==1.0.5
idna==2.6
idna-ssl==1.0.1
Jinja2==2.10
jmespath==0.9.3
keyring==8.7
keyrings.alt==1.3
libnacl==1.6.1
locationsharinglib==0.4.0
MarkupSafe==1.0
multidict==4.1.0
mutagen==1.40.0
netdisco==1.3.1
netifaces==0.10.6
paho-mqtt==1.3.1
pathtools==0.1.2
pkg-resources==0.0.0
protobuf==3.5.2.post1
pushbullet.py==0.11.0
py-vapid==1.3.0
PyChromecast==2.1.0
pycparser==2.18
PyCRC==1.21
pycryptodome==3.4.11
pyfttt==0.3
pyicloud==0.9.1
PyJWT==1.6.0
pylgtv==0.1.7
python-dateutil==2.6.1
python-forecastio==1.4.0
python-jose-cryptodome==1.3.2
python-magic==0.4.15
python-mpd2==1.0.0
python-synology==0.1.0
pytz==2018.3
pywebpush==1.6.0
PyXiaomiGateway==0.9.0
PyYAML==3.12
requests==2.18.4
responses==0.9.0
RestrictedPython==4.0b2
s3transfer==0.1.13
six==1.11.0
speedtest-cli==2.0.0
spotipy==2.4.4
SQLAlchemy==1.2.6
typing==3.6.4
tzlocal==1.5.1
ua-parser==0.7.3
urllib3==1.22
user-agents==1.1.0
voluptuous==0.11.1
voluptuous-serialize==1.0.0
wakeonlan==1.0.0
warrant==0.6.1
watchdog==0.8.3
websocket-client==0.47.0
websockets==3.2
xmltodict==0.11.0
yahooweather==0.10
yarl==1.1.1
zeroconf==0.20.0

```

Nightreaver, 2018-05-08:
If you put the line there... in the correct function, it should definitely _not_ have the same error again!
Please verify... and if its really correct.. we have a false positive of some kind here... will be harder to solve :(

matisaul, 2018-05-08:
I put the line in this fuction:
```
  def check_power(self):
    """Returns the power state of the smart plug."""
    packet = bytearray(16)
    packet[0] = 1
    response = self.send_packet(0x6a, packet)
    err = response[0x22] | (response[0x23] << 8)
    raise SystemExit(type(err))
    if err == 0:
      payload = self.decrypt(bytes(response[0x38:]))
      if type(payload[0x4]) == int:
        if payload[0x4] == 1 or payload[0x4] == 3:
          state = True
        else:
          state = False
      else:
        if ord(payload[0x4]) == 1 or ord(payload[0x4]) == 3:
          state = True
        else:
          state = False
      return state
```
After replacing the file, restarted homeassistant, and... The file backs to original!! The line is been deleted! :/

Nightreaver, 2018-05-09:
Ah, you don't need to restart homeassistant... when you change python code there, its "live"... so you can just change the line... click the button and check the result.
The good thing is, that we might do experiments and you can just revert them by restarting ;)

So please try again without restart

matisaul, 2018-05-09:
I did that, but It's showing the same error :(

Nightreaver, 2018-05-11:
I'm running out of ideas here - I need to setup homeassitant and check whats going on there.
But maybe @mjg59 or @Danielhiversen have an idea about this issue

Danielhiversen, 2018-05-15:
The python3 support is broken for some of the components.
A pr to fix it is welcome.

Nightreaver, 2018-05-15:
i've been testing with py 2.7 and python 3, and it works fine for me... there is no logical issue behind this exception

felipediel, 2021-02-17:
This issue no longer exists. Closing.

closed 2021-02-17 by felipediel