mjg59/python-broadlink #175: RM mini3 devtype /not correctly detected error

WojtekK78, 2018-04-11:
I was trying to detect my newly received RM mini 3, device is properly working from BroadLink app.

Unfortunately rm.devtype is not setup properly with following error via cli:

```
/python-broadlink/cli$ python broadlink_discovery
Discovering...
###########################################
RM2
Traceback (most recent call last):
  File "broadlink_discovery", line 17, in <module>
    print "# broadlink_cli --type {} --host {} --mac {}".format(hex(device.devtype), device.host[0], ''.join(format(x, '02x') for x in device.mac))
AttributeError: rm instance has no attribute 'devtype'
```

When I'm looking at rm object I can see multiple data but not sure if my device is setup correctly (i.e. it's shown as RM2) and definitely devtype is not there

```
{'count': 54236,
 'cs': <socket._socketobject object at 0x7f5e6db97050>,
 'decrypt': <bound method rm.decrypt_pycrypto of <broadlink.rm instance at 0x7f5e6dba3bd8>>,
 'encrypt': <bound method rm.encrypt_pycrypto of <broadlink.rm instance at 0x7f5e6dba3bd8>>,
 'host': ('192.168.0.49', 80),
 'id': '\x02\x00\x00\x00',
 'iv': bytearray(b'V.\x17\x99m\t=(\xdd\xb3\xbaiZ.oX'),
 'key': '\xbb\x00\x9d\n\xf1s\x13z(\xe7\x89i^Z\x00Y',
 'lock': <thread.lock object at 0x7f5e6fdff370>,
 'mac': bytearray(b'\xc0v\x00w\x0fx'),
 'timeout': 10,
 'type': 'RM2'}
```

Any idea how to fix that ? I'm not good with python so any help would be usefoul.

Nightreaver, 2018-05-08:
I am not sure, it looks like you use two different version of this library

The latest breadlink_discovery.py needs and up-to-date __init__.py, otherwise it will fail

make sure the __init__.py you are using looks like this in the first `dev gendevice`

```
  if devtype == 0: # SP1
    return sp1(host=host, mac=mac, devtype=devtype)
  elif devtype == 0x2711: # SP2
    return sp2(host=host, mac=mac, devtype=devtype)
```

if there is `devtype=devtype` missing you are probably using the wrong version

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel