mjg59/python-broadlink #176: Pyaes encryption for S1C

lily148, 2018-04-12:
Hi,

I notice the S1C code is using only Crypto to decrypt bytes.
I try to convert to pyaes but it fail. Anyone have idea?

- Original
aes = AES.new(bytes(self.key), AES.MODE_CBC, bytes(self.iv))
- My code for pyaes       
aes = pyaes.AESModeOfOperationCBC(self.key, iv = bytes(self.iv))

- Original
payload = aes.decrypt(bytes(response[0x38:]))
- Original
payload = b"".join([aes.decrypt(bytes(response[i:i+16])) for i in range(0, len(response[0x38:]), 16)])

I am sure somewhere I miss but don't know where.....
Anyone can help?

Nightreaver, 2018-05-08:
We already have some pyaes implementation here (https://github.com/mjg59/python-broadlink/blob/master/broadlink/__init__.py#L178)

Maybe you can check there?

And if you commit, try similar approach with supporting both ... pyaes and crypto

felipediel, 2021-02-17:
Closing, since the problem cannot be fixed in our code base.

closed 2021-02-17 by felipediel