mjg59/python-broadlink #181: [Question] how can I send combo command?

vajonam, 2018-04-25:
just call the CLI multiple times? is there any other trick?

Nightreaver, 2018-05-08:
Can you explain a bit futher ? What are you trying to achieve?

vajonam, 2018-05-08:
Trying to send a channel number like 7-8-0-OK

vajonam, 2018-05-13:
> If you don't want to send multiple commands for a frequently used
> combination of buttons, try hitting the multiple buttons in the learning
> mode as one command.

No it only learns one button at a time.

Nightreaver, 2018-05-14:
Yes i can tell it doesn't learn multiple keys at once, I ended up learning 0-9 and OK keys and send them one by one in order to change the channel... its slow but worked for my TV

kovikrisz, 2018-05-17:
Any suggestion how could I setup sleep timer (send 3x same command within a shorter time frame).
In case I use CLI send command, the gaps between the data sending are too big for my TV to recognize (will go back always to the main menu due to timeout)

uudruid74, 2018-06-12:
I implemented Macros in my tree.  When I say good morning to Google Assistant it fires off a macro that turns on the TV, changes the temp setting on the AC, then sends down/right/ok combinations to the TV to work the menus to get the news playing.  The macros run fast.

If there is interest, I could add whatever @mjg59 has added here into my tree and issue a PR?

batinthestacks, 2018-06-13:
uudruid74, how are you getting Google Assistant to call your HTTP server?

kovikrisz, 2018-06-17:
uudruid74, could you please share your macro files?

uudruid74, 2018-06-17:
It's implemented my fork of @radinky's http-rest server which calls this project as a library.  Being a long-running server means no time to setup and tear down a new Python process for each command, and its easy to build a remote using simple Ajax calls.  Google Assistant integration is done via IFTTT and the "WebHooks" service, basically you select the text and what URL to call.  The basic/stable version is quite good and now part of @radinsky's tree.  I'm working on a much more powerful version as well (working, stability unknown) in an advanced-macros branch.

Feel free to discuss there: https://github.com/uudruid74/broadlink-http-rest
Demo: https://youtu.be/ysfbJsk43Vk

closed 2024-04-21 by felipediel