mjg59/python-broadlink #184: TypeError: __init__() takes exactly 4 arguments (3 given)

felinh, 2018-05-01:
Hello
Scanning network for Broadlink devices
but after calling  

#!/usr/bin/python
 import broadlink
import time
import sys 
device = broadlink.rm(host=("xxx.xxx.xxx.xxx",80), mac=bytearray.fromhex("xxxxxxx"))
--
console displays
**TypeError: __init__() takes exactly 4 arguments (3 given)**

what is going wrong?

andrez7, 2018-05-06:
Hi Felinh 

Try with broadlink.rm(IPAddress, Port, MACAddress, Timeout)
In my project I added the 4th arguments and It works. p.e Timeout = 30

AntonLevi, 2018-05-22:
import broadlink
import time
import sys
ip = "192.168.1.33"
port = 80
mac = "b44345455555"
timeout = "30"
device = broadlink.rm((ip,port), mac, timeout)

felinh, 2018-05-22:
thanks guys, I do confim the timeout argument was missing.
It works now.
PS: source [here](http://domo-attitude.fr/broadlink-rm-promini-domoticz/) to be updated.

closed 2018-05-22 by felinh

albebert, 2018-10-11:
> 
> 
> thanks guys, I do confim the timeout argument was missing.
> It works now.
> PS: source [here](http://domo-attitude.fr/broadlink-rm-promini-domoticz/) to be updated.

Hello, how have you added the timeout arugment please ? tried some things but no luck :( 

thanks