mjg59/python-broadlink #186: Learn mode: No data received

Komagain, 2018-05-07:
Hello
Each time I try to make my RM learn (IR or RF), the following message is displayed:
```
Learning
No data received
```
I am able to learn via ihc app
What can I check ?

Thanks

Nightreaver, 2018-05-08:
What are you trying to learn? IR ?

Komagain, 2018-05-08:
I have tried both IR and RF
It works with the broadlink app.

rwa, 2019-01-07:
I am having the same problem, only with RF devices.  I can learn them on the ihc app, but I always time out trying to learn them through the cli.  (I can learn IR codes just fine).

My device type is being reported as 0x279d, which seems to be new.  Is it possible the protocol has been changed for this device?

vchoi, 2019-07-09:
I'm having the same issue while trying to learn IR codes. Device 0x2787

[image]

mtronicsDev, 2019-10-20:
> I am having the same problem, only with RF devices. I can learn them on the ihc app, but I always time out trying to learn them through the cli. (I can learn IR codes just fine).
>
> My device type is [...]

I have the same device type and have been able to learn RF exactly once. Everytime apart from the one where it worked, I always got past the RF sweeping part but then, no code was found.

The one time it worked, I got a code which I could send without a problem.

zero1niner, 2020-11-01:
Having the same problem

zero1niner, 2020-11-01:
wait..I found a solution in my situation. Hoping my hours of pain will help someone else. 

Someone gave me this tip, which opens the help doc
./broadlink_cli -h

From there, I saw the --rfscanlearn command. 

Changed my string to have the --rfscanlearn at the end instead of the learn i had been using, and suddenly the thing started learning my remotes. 

I have spent DAYS on this, and if I were only a better developer, I would commit to writing up a full how to that is complete and current. But im not clear enough on the steps, and I would only confuse people more by trying. 

For RF devices, I changed the 

./broadlink_cli --type 0x649b --host 192.168.1.142 --mac a043b018f07b --rfscanlearn

felipediel, 2021-03-12:
Case closed. I will improve the documentation to inform about `--rfscanlearn`.

closed 2021-03-12 by felipediel