mjg59/python-broadlink #188: rm instance has no attribute 'set_power'

digracesion, 2018-05-10:
i'm running the command `python broadlink_cli --device "(rm type) (host) (mac)" --turnon` to turn on my television but command prompt kept producing  the message:

```
Traceback (most recent call last):
File "broadlink_cli", line 154, in <module>
print dev.set_power(True)
AttributeError: rm instance has no attribute set_power

```
the `--learnfile` and --send commands are working though, and it says here in the broadlink_cli file that `--turnon` command is for turning on the device:

`parser.add_argument("--turnon", action="store_true", help="turn on device")`

but it doesn't specify which type of device, unlike the line for turnonnlon (which is for nightlight)
is it because the command isn't compatible with the television? or is part of my input command wrong?

Nightreaver, 2018-05-11:
Can you please give the whole command you are using ? you can replace ip/mac if you want to

digracesion, 2018-05-14:
i'm running it as `python broadlink_cli --device "0x272a 192.168.X.X XXXXXXXXXXX" --turnon`

Nightreaver, 2018-05-14:
Hey...

I think you misunderstood the use of the command.
RMpro can turn on/off your TV, but only when you **LEARN** the correct command first, and if you did, you need to send that command back via `--send` to RMpro to turn the TV on or off or do any other IR commands.

The `--turnon` and `--turnoff` ... etc commands are used by power adapters, RMpro cant make use of those.

digracesion, 2018-05-14:
oh, i see. i thought they were commands that can be used in the RMPro, which made me confused since i was able to learn the commands on my remote well despite the command not working. thanks!

Nightreaver, 2018-05-14:
Yeah, we should add a line that this command is not supported rather then having it printing out an error...

closed 2018-07-03 by digracesion