mjg59/python-broadlink #191: discover on multihomed hardware fails

mcc05, 2018-05-22:
if the broadlink discovery is run on a machine with say two interfaces then it fails even if the localipaddress is specified and this does not match the primary network. An option to bind to a particular network interface would be useful or the ability to manually specify devices and ip/mac and not use discovery

sbsrouteur, 2018-05-22:
I had the same problem and suggested PR# 173 to fix it, could you try it on your setup ?

mcc05, 2018-05-22:
Hi yep I'll give it a go at weekend sorry for the newbie python question can I just not the init.py or something more complicated ?

mcc05, 2018-05-22:
mod that is ?

sbsrouteur, 2018-05-22:
The init.py change is all that is required. about 10 lines of code.

mcc05, 2018-05-25:
@sbsrouteur, it seems to work great, only thing should it not include the same approach for the last line of the code for the function? I would support this change :)

# Setup a new Broadlink device via AP Mode. Review the README to see how to enter AP Mode.
# Only tested with Broadlink RM3 Mini (Blackbean)
def setup(ssid, password, security_mode):

mcc05, 2018-05-25:
@sbsrouteur I've now tried the code on a RasPi with both ethernet and wifi, running Hassio, and built it into a docker container, then code runs ok, but can't get discovery to work :( 
I know the docker container has access to three networks, the internal, the ethernet and the wifi, but even with using the new broadcast_ip_address nothing is found. If I manually specify a device in code then the rest of the code works? Any ideas?

mcc05, 2018-05-26:
@sbsrouteur The networks are 192.168.0.0 for the ethernet, 192.168.10.0 for the devices including the broadlinks ones, and there is the internal 172x address internal address for the docker containers. I am setting the broadcast IP as 192.168.10.255

sbsrouteur, 2018-05-26:
This sound good, and did you put you local ip on the 192.168.10.x net as well ?
All these nets have 255.255.255.0 netmasks?

For me with a 192.168.99.x LAN and 192.168.100.x Wifi, with the  broadlink on the wifi, I use this command for discovery : 
broadlink.discover(timeout=1,local_ip_address='192.168.100.1',broadcast_ip_address='192.168.100.255')

mcc05, 2018-05-26:
@sbsrouteur Thanks for the help I figured it out it was to do with the docker container build, I had to set an option to access the host network, all works fine now.

Tuipveus, 2018-12-21:
what is that broadcast_ip_address and where it is defined ? New broadlink.discover does not seem to have it, which of the older has it if any ?
What google dns 8.8.8.8 has anything to do with what ???

Tuipveus, 2018-12-28:
Thanks of your reply sbsrouteur.
Actually I tried to set local_ip_address with two different computers (laptop and rpi) and with laptop it started to work fine, but with RPI not (even used the same networks). Then I tried PR173, but it still didn't work by setting broadcast_ip_address to 192.168.2.255 or 255.255.255.0. Eventually I got it working, by setting local_ip_address to 192.168.2.35 (which is the RPI) and broadcast_ip_address to 192.168.2.33 (which is the device that I want to control). I don't have any gateway in that network (but I didn't have any gateway when I used laptop too). I hope this can help someone with the same problem.

So now it finally works like it should, just have to make sure that both RPI and device get old addresses from dhcp everytime.

Why main branch (code without PR173) worked with laptop, but not with rpi, that still is mystery to me, but it does not matter as long as it works.

sbsrouteur, 2018-12-29:
To answer the question regarding why the RPI does not work is the broadcast address is not used. It is RPI specific. When binding to the broadcast address to listen for retuning packets, if you have more than 1 NIC, it only listens on the first one, unless you give a mask that forces listening on the other one.

felipediel, 2021-04-06:
Fixed with https://github.com/mjg59/python-broadlink/pull/427.

closed 2021-04-06 by felipediel