mjg59/python-broadlink #192: Help encoding IR into Broadlink Format

newzealandpaul, 2018-05-29:
I am trying to use the HVAC-IR-Control control project to generate IR codes for my heatpump, beyond the limited functionality the Broadlink database is capable of.

The IR codes dumps from the before mentioned projects is 0's and 1's (which I could convert to binary, base64 or whatever).

For example: 

> 10 100000 11100000 100 0 1001000 110010 10000000 10101111 0 0 1110 11100000 10000 0 1 0 110 10110100 .

More codes here:

https://gist.github.com/newzealandpaul/25f2da6b49ce3dcc50c55c5ee96b1850

Does anyone know how I can convert this to the Broadlink format. I have looked at the various posts here but I cannot work out how I could transform into a broadlink packet. 

Would greatly appreciate some help on this.

Nightreaver, 2018-05-29:
Hey Paul

you can use to the cli to convert "duration" format to broadlink format and vice versa with "convert/duration" command line option.

can you get the "duration format" from hvac-ir-vr project?

cheers

newzealandpaul, 2018-05-29:
Hey,

It does not, it not really designed to output IR rather than control an IR sensor directly via GPIO pins, but the project is well documented.

If durations are the duration in microseconds of the IR being on and off, then I think I can hack it to do that. 

Can you give me an example of the duration format? Its not quite clear to me from the cli code.

Nightreaver, 2018-05-29:
I just found that one http://irdb.tk/convert/ as sample

you can see "Raw" format on to which is the format you can use with broadlink.
Maybe you can use that site already to convert?

Otherwise I can't really tell you how to get there from binary or your hex format (looks incomplete for me)...

Just found that on my search - http://www.righto.com/2010/03/understanding-sony-ir-remote-codes-lirc.html

there might be better help sources out there....

newzealandpaul, 2018-05-30:
Thanks Nighreaver. I have tried to generate some. Am I using it right?

./broadlink_cli --type 0x272a --host xxxxx --mac xxxxx --durations "-0 +3500 -1750 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -10000 +3500 -1750 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -1300 +435 -1300 +435 -1300 +435 -435 +435 -1300 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -1300 +435 -435 +435 -435 +435 -435 +435 -435 +435 -435 +435 -1300 +435 -435 +435 -1300 +435 -435 +435 -1300 +435 -1300 +435 -435 +435 -1300 +435 -10000 -0"

Nightreaver, 2018-05-30:
hey

yeah the format looks okay.

regarding usage... i think "--durations" is just to convert, then you need to use another option "send" or so to actually send the command to broadlink

so it's better to save the broadlink format to use it frequently

Nightreaver, 2018-05-30:
> @Nightreaver <https://github.com/Nightreaver> thanks. I tried the
> --durations --send command and it seems to send it (LED lights up). Its
> still not doing anything to my heatpump so I am at a complete. I must be
> generating bad durations.
>
> Such a pity, I feel I am so close!!!
>
> Thanks anyway, I really appreciate all your help :)

Yeah, it looks right to me as well, maybe just some wrong timings.

But as I never really used durations... I cant really help with that :(

closed 2024-04-21 by felipediel