mjg59/python-broadlink #194: CLI Send Command (from File) Not Working

Digicrat, 2018-06-12:
I just got the RM3 and am having multiple issues setting it up, but starting with the basics, I've joined the RM3 to the network (via broadlink.setup() function since the SURE app isn't working for me).

I then tried using the provided cli script:
- ./broadlink_discover    # On the third try (most attempts fail), it found the device and I output the line to a file
- ./broadlink_cli --device @loft.device --learnfile JVC.VolDown  # This worked
- ./broadlink_cli --device @loft.device --send @JVC.VolDown   # This does not
- ./broadlink_cli --device @loft.device --send 2600940000011589122331... # Contents of saved file (truncated here for brevity).  This gives no errors, but doesn't cause my volume to change either.

The send command from file fails with an "error: unrecognized arguments: " message that echoes the contents of my JVC.VolDown file.  Including the contents of the file in the commandline gives no error, but has no apparent effect either (I'll need to setup another IR receiver later to check whether it's transmitting anything).  

Am I missing something here?

Also, the discovery (when it works) reports RM2 and not RM3; is that significant?

Thanks

darkpainy, 2018-08-31:
I have the same issue.
I can learn all the commands, but I can't send them from the file.
broadlink_cli: error: unrecognized arguments: 2600840026132513261522130c2b0c2c0d2b0d2b0b2c25130c2b0b2d0c2c26120d2b25122715050a141522120c2c2711261b1c130c2b261226130b2c0c2b26120d2b0c2c250003d3251a1c14251424130c2b0c2c0c2b0d2b0c2c264a0b2c0c2c25130c2b26132512261226120c2c2513241425130b2c261226120c2c0c2b26120d2b0c2c25000d0500000000

ciberado, 2018-10-26:
As daiyambal said the device file should contain something like "0x2737 192.168.1.34 0bb547770f78", without any CR/LF at the end of the line.

alpet83, 2020-09-24:
Same problem I solved, via change code in broadlink_cli:
-parser.add_argument("data", nargs='', help="Data to send or convert")
+parser.add_argument("--data", nargs='', help="Data to send or convert")
Now it works perfectly: broadlink_cli --device @ac --send -data @t24

darkpainy, 2020-09-24:
> Same problem I solved, via change code in broadlink_cli:
> -parser.add_argument("data", nargs='', help="Data to send or convert")
> +parser.add_argument("--data", nargs='', help="Data to send or convert")
> Now it works perfectly: broadlink_cli --device @ac --send -data @t24

@alpet83 you mad genious 👍 who knew it would be as easy as adding two missing -'s.  
Thanks man.

Have a nice evening.