mjg59/python-broadlink #198: RF on RM pro+ ?

okaapi, 2018-06-30:
This is nice! I can learn IR commands on my RM pro+   with   .enter_learning() and .check_data()

But it cannot get it to learn RF commands?

mob41, 2018-07-03:
To receive RF commands, “sweep frequency” command is required. However, it is currently not implemented in this library. Therefore, you can only receive IR commands but not RF commands.

There’s a workout for this. You can receive the RF data from a third party like RMBridge and hook the data to the send command. 

The string RF data is just a bunch of HEX code which you can deal with <code>b”173e2748290”</code> to convert to byte array.

https://github.com/mjg59/python-broadlink/issues/87

okaapi, 2018-07-11:
Solved...
[https://github.com/mjg59/python-broadlink/issues/87](url)

closed 2018-07-11 by okaapi