mjg59/python-broadlink #200: New Device: Efergy ego Smart Plug

ecylcje, 2018-07-27:
Hi,

By adding the device type:

0x271d                           # Efergy ego

I have managed to get the Efergy ego smart plug to be switchable (on / off). As group SP2.

Theoretically it should also be able to report back power usage but this does not seem to work when added to the SP2 group. I just get  a 'None' returned.

print devices[0].get_energy()
None

I'm afraid I don't know how to debug the returned values? Is there a way I view what data is returned?

Thanks,

Chris

ecylcje, 2018-08-01:
I managed to get this working by changing the packet details to send all 0s, plus I needed to change the calculation:

  def get_energy(self):
    packet = bytearray(16)
    packet[0] = 4
    response = self.send_packet(0x6a, packet)
    err = response[0x22] | (response[0x23] << 8)
    if err == 0:
      payload = self.decrypt(bytes(response[0x38:]))
      if type(payload[0x05]) == int:
        energy = (float(hex(payload[0x06]) * 256 + float(payload[0x05]) + float(hex(payload[0x04])))/100.0)/3.9093
      else:
        energy = (float(ord(payload[0x06]))*256 + float(ord(payload[0x05])) + float(ord(payload[0x04]))/100.0)/3.9093
      return "%.1f" % energy

I have commited the changed to my github repo:

https://github.com/ecylcje/python-broadlink

while I try and work out how to get the daily/weekly/yearly results from the device. I believe if you send the packet[0]=5 then the response contains all the data. I just don't know understand everything enough.

Chris

maartentromp, 2018-08-29:
Thanks a lot, it works on my Mac! I will now try to get it to work in Domoticz on my RapberryPi.
Regards, Maarten

tottka, 2019-01-10:
Thank you so much for posting your code.  I am trying to decode a ANKUOO Neo Pro and your code got me 1 step closer.  I can now tell if electricity is being utilized or not but the calculation currently needs work.   Any hints on how you solved the calculation?

andreaslordos, 2019-11-30:
Hi, @ecylcje do you mind providing more details/steps on how to connect the Efergy Smart Plug to a device like the Raspberry Pi? Tried following the README but it didn't work

ecylcje, 2019-12-02:
> Hi, @ecylcje [1] do you mind providing more details/steps on how to
> Links:

Hi,

I'm just using something similar to:

d1 = broadlink.efergy(host=("192.168.92.116",80), 
mac=bytearray.fromhex("b4 43 0d ee c5 1c"),devtype="0x271d")

and then

power = d1.get_energy()

the only note is that I'm not sure whether the efergy code made it into 
Master, otherwise the code is:

https://github.com/ecylcje/python-broadlink

Chris

andreaslordos, 2019-12-02:
Thank you! I managed to figure it out eventually using broadlink.discover()

I just wrongly assumed the Efergy socket had to be in AP mode - instead,
the Efergy had to already be set up on the network via the Ego app on my
phone and had to be in it's normal (idle) state for it to be picked up by
broadlink.discover()

Andreas

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/481. Thank you!

closed 2021-02-17 by felipediel