mjg59/python-broadlink #201: "Higher level" send_data() implementation

lbschenkel, 2018-08-02:
Would you be open for this library to implement a new method, equivalent to `send_data()`, in which the argument is a high-level description of the data to be sent? One example would be to be able to send a IR packet described in Pronto hex format:

```
convert_and_send("pronto_hex", "0000 0070 0000 0032 0080...")
```

This method would simply convert the higher level representation into the Broadlink raw bytes and pass those to `send_data()`.

The motivation for this issue is PR https://github.com/home-assistant/home-assistant/pull/15796.

mglatz, 2019-07-29:
Id love this too. Id like to be able to send commands I download from net (such as http://www.remotecentral.com/cgi-bin/codes/samsung/tv_functions/) but I have no clue how...

lbschenkel, 2019-08-18:
> Id love this too. Id like to be able to send commands I download from net (such as http://www.remotecentral.com/cgi-bin/codes/samsung/tv_functions/) but I have no clue how...

You might be interested in https://github.com/lbschenkel/broadlink-bridge

closed 2022-07-01 by lbschenkel