mjg59/python-broadlink #220: no devices in devices list after install the plugin in Domoticz

stillyu, 2018-12-07:
I installed the plugin according to https://www.domoticz.com/wiki/Plugins/BroadlinkRM2.html
After i add the hardware in Domoticz, there is no new device in devices list.

The logs:
2018-12-08 01:27:19.015 Status: PluginSystem: Started, Python version '3.5.2'.
2018-12-08 01:27:19.037 Status: WebServer(HTTP) started on address: :: with port 8080
2018-12-08 01:27:19.041 Status: WebServer(SSL) started on address: :: with port 443
2018-12-08 01:27:19.042 Status: Proxymanager started.
2018-12-08 01:27:19.043 Status: TCPServer: shared server started...
2018-12-08 01:27:19.043 Status: RxQueue: queue worker started...
2018-12-08 01:27:21.043 Status: (Broadlink) Started.
2018-12-08 01:27:21.044 Status: EventSystem: reset all events...
2018-12-08 01:27:21.046 Status: EventSystem: reset all device statuses...
2018-12-08 01:27:21.068 Status: Python EventSystem: Initalizing event module.
2018-12-08 01:27:21.068 Status: EventSystem: Started
2018-12-08 01:27:21.068 Status: EventSystem: Queue thread started...
2018-12-08 01:27:21.260 Status: PluginSystem: Entering work loop.
2018-12-08 01:27:22.158 Status: (Broadlink) Entering work loop.
2018-12-08 01:27:22.158 Status: (Broadlink) Initialized version 4.1.0, author 'zak45'

OS: Ubuntu 16.04
Domoticz version: 4.9701

stillyu, 2018-12-07:
pip3 uninstall pycryptodome
solve the problem

closed 2018-12-07 by stillyu

mgenrique, 2019-01-01:
Thanks, It solves for me too.