mjg59/python-broadlink #223: Error in hysen device, send request, CRC16

uspass, 2018-12-28:
Symptom:
Two hysen devices are set.
For the second device, most of the times, an error is raised by:
crc = CRC16(modbus_flag=True).calculate(bytes(input_payload))
The crc is None 
The error is:
 unsupported operand type(s) for &: 'NoneType' and 'int'
Workaround:
The line:
from PyCRC.CRC16 import CRC16
should be taken outside the class hysen.

Cheers,
us

ptd006, 2019-03-02:
Hello uspass,

thanks for pointing this out. 

I wrote it but never noticed any issue because I only had 1 device.

Please submit a patch if you get chance.

Cheers,
Peter

felipediel, 2021-02-17:
We no longer use this import, so I think this is fixed. If not, please reopen the issue.