mjg59/python-broadlink #230: Domoticz don't find Broadlink devices

ArPaQua, 2019-02-11:
Hi everyone,
my Domoticz system don't find any broadlink device.
I installed one RM2 with 3 devices: air cond, TV and Bose speaker. It works with the App (ihc For EU) but in Domoticz he can't find the devices.
IP and MAC are corrects, below the log:
`2019-02-11 18:20:50.692 (Broadlink RM2) Debug logging mask set to: PYTHON PLUGIN QUEUE IMAGE DEVICE CONNECTION MESSAGE ALL
2019-02-11 18:20:50.692 (Broadlink RM2) Creating device 'Command'.
2019-02-11 18:20:50.692 (Broadlink RM2) Creating device 'Remote'.
2019-02-11 18:20:50.692 (Broadlink RM2) 'Name':'Broadlink RM2'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Address':'192.168.1.15'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Mode3':'RM2'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Port':'0'
2019-02-11 18:20:50.692 (Broadlink RM2) 'HardwareID':'7'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Mode4':'no'
2019-02-11 18:20:50.692 (Broadlink RM2) 'DomoticzVersion':'4.10426'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Key':'BroadlinkRM2'
2019-02-11 18:20:50.692 (Broadlink RM2) 'DomoticzHash':'91a981ef'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Mode2':'/log/BroadlinkRM2'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Database':'/home/pi/domoticz/domoticz.db'
2019-02-11 18:20:50.692 (Broadlink RM2) 'Language':'en'
2019-02-11 18:20:50.693 (Broadlink RM2) 'Version':'4.1.0'
2019-02-11 18:20:50.693 (Broadlink RM2) 'DomoticzBuildTime':'2019-02-10 12:51:47'
2019-02-11 18:20:50.693 (Broadlink RM2) 'Mode6':'Debug'
2019-02-11 18:20:50.693 (Broadlink RM2) 'UserDataFolder':'/home/pi/domoticz/'
2019-02-11 18:20:50.693 (Broadlink RM2) 'HomeFolder':'/home/pi/domoticz/plugins/BroadlinkRM2/'
2019-02-11 18:20:50.693 (Broadlink RM2) 'Mode5':'9000'
2019-02-11 18:20:50.693 (Broadlink RM2) 'Author':'zak45'
2019-02-11 18:20:50.693 (Broadlink RM2) 'Mode1':'780F77FD3BB0'
2019-02-11 18:20:50.693 (Broadlink RM2) 'StartupFolder':'/home/pi/domoticz/'
2019-02-11 18:20:50.693 (Broadlink RM2) Device count: 0
2019-02-11 18:20:50.693 (Broadlink RM2) Pushing 'PollIntervalDirective' on to queue
2019-02-11 18:20:50.693 (Broadlink RM2) Connecting to: 192.168.1.15:780F77FD3BB0
2019-02-11 18:20:50.800 (Broadlink RM2) Connected to Broadlink device: 192.168.1.15
2019-02-11 18:20:50.822 (Broadlink RM2) No ini file :/log/BroadlinkRM2/remote/plugin_remote_7.ini
2019-02-11 18:20:50.822 (Broadlink RM2) Custom Commands for Remote not managed
2019-02-11 18:20:50.822 (Broadlink RM2) Device Number begin to : -1
2019-02-11 18:20:50.822 (Broadlink RM2) Processing 'PollIntervalDirective' message
2019-02-11 18:20:50.822 (Broadlink RM2) Heartbeat interval set to: 30.`
How can I fix it?
thank you

ArPaQua, 2019-02-13:
With the help of Zak now I have this situation:

> 2019-02-13 10:06:14.782 (Broadlink RM2) Creating device 'Import'.
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Key':'BroadlinkRM2'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Version':'4.1.0'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Port':'0'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'HomeFolder':'/home/pi/domoticz/plugins/BroadlinkRM2/'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Address':'192.168.1.15'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Language':'en'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Mode6':'Debug'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Name':'Broadlink RM2'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'DomoticzBuildTime':'2019-02-12 11:11:30'
> 2019-02-13 10:06:14.783 (Broadlink RM2) 'Mode3':'RM2'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'DomoticzHash':'75069fe9'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Mode1':'780F77FD3BB0'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'DomoticzVersion':'4.10427'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Mode5':'9000'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'StartupFolder':'/home/pi/domoticz/'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'HardwareID':'8'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Mode2':'home/pi/domoticz/log/BroadlinkRM2'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Database':'/home/pi/domoticz/domoticz.db'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'UserDataFolder':'/home/pi/domoticz/'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Mode4':'yes'
> 2019-02-13 10:06:14.784 (Broadlink RM2) 'Author':'zak45'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device count: 3
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device: 1 - ID: 35, Name: 'Broadlink RM2 - Command', nValue: 0, sValue: ''
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device ID: '35'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device Name: 'Broadlink RM2 - Command'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device nValue: 0
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device sValue: ''
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device LastLevel: 0
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device: 254 - ID: 36, Name: 'Broadlink RM2 - Remote', nValue: 0, sValue: ''
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device ID: '36'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device Name: 'Broadlink RM2 - Remote'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device nValue: 0
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device sValue: ''
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device LastLevel: 0
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device: 255 - ID: 37, Name: 'Broadlink RM2 - Import', nValue: 0, sValue: ''
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device ID: '37'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device Name: 'Broadlink RM2 - Import'
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device nValue: 0
> 2019-02-13 10:06:14.784 (Broadlink RM2) Device sValue: ''
> 2019-02-13 10:06:14.785 (Broadlink RM2) Device LastLevel: 0
> 2019-02-13 10:06:14.785 (Broadlink RM2) Pushing 'PollIntervalDirective' on to queue
> 2019-02-13 10:06:14.785 (Broadlink RM2 - Import) Updating device from 0:'' to have values 0:'Off'.
> 2019-02-13 10:06:14.790 (Broadlink RM2) Update 0:'Off' (Broadlink RM2 - Import)
> 2019-02-13 10:06:14.790 (Broadlink RM2) Connecting to: 192.168.1.15:780F77FD3BB0
> 2019-02-13 10:06:14.854 (Broadlink RM2) Connected to Broadlink device: 192.168.1.15
> 2019-02-13 10:06:14.854 (Broadlink RM2 - Command) Updating device from 0:'' to have values 1:'10'.
> 2019-02-13 10:06:14.859 (Broadlink RM2) Update 1:'10' (Broadlink RM2 - Command)
> 2019-02-13 10:06:14.862 (Broadlink RM2) No ini file :home/pi/domoticz/log/BroadlinkRM2/remote/plugin_remote_8.ini
> 2019-02-13 10:06:14.862 (Broadlink RM2) Custom Commands for Remote not managed
> 2019-02-13 10:06:14.862 (Broadlink RM2) Device Number begin to : 1
> 2019-02-13 10:06:14.863 (Broadlink RM2) Processing 'PollIntervalDirective' message
> 2019-02-13 10:06:14.863 (Broadlink RM2) Heartbeat interval set to: 30.
> 2019-02-13 10:06:14.776 Status: (Broadlink RM2) Entering work loop.
> 2019-02-13 10:06:14.777 Status: (Broadlink RM2) Initialized version 4.1.0, author 'zak45'
> 2019-02-13 10:06:24.778 (Broadlink RM2) Pushing 'onHeartbeatCallback' on to queue
> 2019-02-13 10:06:24.779 (Broadlink RM2) Processing 'onHeartbeatCallback' message
> 2019-02-13 10:06:24.779 (Broadlink RM2) Calling message handler 'onHeartbeat'.

The problem is that I find nothing in Switch tab

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel