mjg59/python-broadlink #235: RM3 mini password length limit?

thatso, 2019-04-14:
After failing to connect a recently purchased Broadlink RM 3 mini (aka Black Bean) to my WLAN with all three Android apps (e-control, Intelligent Home Center, Intelligent Home Center EU) even in AP mode because apparently the glorious Broadlink engineers randomly and against the rules chose to limit the password length to 32 characters, I still wonder if this is a software or hardware limit.
While the apps constantly and silently failed to connect, I received only once a hint that the WLAN password is limited to 32 characters.
Trying with the latest python-broadlink, I get this error:

```
>>> import broadlink
>>> broadlink.setup('ssidofmynetwork', 'mylongnetworkpassword', 3)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "broadlink/__init__.py", line 865, in setup
    payload[(pass_start + pass_length)] = ord(letter)
IndexError: bytearray index out of range
```

Needless to say that an attempt with a shorter password succeeded.
I hoped that using python-broadlink instead of the vendors' apps might overcome this limitation. Now I'm afraid its an annoying hardware limit.

drew-kun, 2020-11-09:
Thanks for this.. you saved me a lot of time trying to find out the max limit, as my wifi networks use 63 char wpa passphases.

KTibow, 2020-11-09:
So should this throw an exception then?

drew-kun, 2020-11-11:
Ideally, should support max wpa char length, but if no, then at least throw out some sensible warning/error/exception... Because for now these devices are the only reason I use 32 char passphrase instead of 63, meaning that concept of using broadlink ir controllers lowers the security level of the wireless network, which is not good...

KTibow, 2020-11-11:
@drew-kun 32 chars is very very secure. Take for example `3h2i6u6LaeX8AAP` (15 chars, not my real password) and run it through a password checker you can find on Google.

felipediel, 2021-02-17:
You need to update the firmware, this limitation no longer exists in the new version.

Closing, since the problem is not related to our code base. But adding this to a troubleshooting section in the documentation wouldn't be such a bad idea.

closed 2021-02-17 by felipediel

IgitBuh, 2021-10-05:
> 
> You can try to update the firmware, I think this limitation no longer exists in the new version.
> 

Is there some kind of firmware changelog available somewhere?

My RM4 Pro came with v52078 and I am unable to use a WPA2 passphrase longer than 32 chars.
The official app shows that there is a firmware update available. However, the as soon as it loads to 100%, an error tells me: "Updating is failed". If a newer firmware version solves this issue, then I would try to find a solution for updating it.

In case the latest firmware does not solve this issue, is there maybe a way around this?
Wifi WPA/2 encryption is based on a PBKDF2 conversion of the SSID and the passphrase. The actual security key is a 64 hex digits value. Maybe it is possible to send this locally converted value to the Broadlink device instead of the unencrypted passphrase, circumventing the passphrase length limitation?

Steltek, 2021-10-05:
> Maybe it is possible to send this locally converted value to the Broadlink device instead of the unencrypted passphrase, circumventing the passphrase length limitation?

The limitation is in the protocol used to send the configuration (and possible in storage too) so there's no way to send a longer password without a change to the firmware. (I worked around the issue by ripping out the original control board of my RM3 and replacing it with a Wemos D1 running Tasmota: https://github.com/chaeplin/rm-mini3-to-esp8266/issues/2#issuecomment-517936287 . If you can still return yours, it's probably cheapest to do that and get a Wemos D1 and some IR send/receive circuit bits instead.)

IgitBuh, 2021-10-05:
> 
> If you can still return yours, it's probably cheapest to do that and get a Wemos D1 and some IR send/receive circuit bits instead.)

Well, I can still return it, but I'm not sure if that's really an alternative. I purchased the RM4 Pro specifically for being able to control both IR and RF devices. I understand that there is the possibility of building a custom IR/RF control device myself, but I wasn't planning on investing days or weeks into this project. If there is a good alternative IR+RF device with openHAB support, I'd be more than happy to hear about it!