mjg59/python-broadlink #237: New broadlink bulb from Clas Ohlson

vstoms, 2019-04-26:
Hi,
Clas Ohlshon is selling an smart Wifi bulb thats I think is an broadlink device.

Norway: https://www.clasohlson.com/no/Clas-Ohlson-Home,-WiFi-Smart-Bulb,-lysp%C3%A6re/38-8774
Sweden: https://www.clasohlson.com/se/WiFi-Smart-Bulb-Clas-Ohlson-Home/38-8774
German: https://www.clasohlson.com/de/WLAN-Smart-LED-Clas-Ohlson-Home/38-8774

IHC app is calling this device SB800TD

Data from the scripts:

broadlink_discovery
###########################################
Unknown
# broadlink_cli --type 0x5043 --host 10.0.0.XX --mac 9efbd6770f78
Device file data (to be used with --device @filename in broadlink_cli) : 
0x5043 10.0.0.XX 9efbd6770f78

###########################################

from broadlink:

<broadlink.device instance at 0x7fa5e4138200>

My task is to get these bulbs into home-assistant, so what should be the next step?

nihilus, 2019-11-21:
Any updates on this?

nihilus, 2019-11-25:
Trying to treat these as 'SP2' and got an empty payload back (after decrypting it, looks like it's padded for AES).

skrutt, 2019-12-03:
Did you succeed on this?

nihilus, 2020-04-23:
@skrutt Yes, it's a variant of the SP4 protocol.

kuba160, 2020-06-13:
@nihilus any updates on this? I got this bulb and would like to control it through cli. I don't have a clear view what you mean with prepending length to 16 bytes, could you explain it again?

nihilus, 2020-06-13:
> @nihilus any updates on this? I got this bulb and would like to control it through cli. I don't have a clear view what you mean with prepending length to 16 bytes, could you explain it again?

I did a working solution in C for my former employer which I cannot share due to being intellectual property beside not having access to the source code anymore.

But if you sniff packets from the app and look at the packet format being implemented it should be straightforward to implement.

kuba160, 2020-06-13:
I just noticed somebody started working on it in https://github.com/mjg59/python-broadlink/pull/353.
After some fiddling I managed to turn the lightbulb on/off 😄. Thanks for quick answer anyway.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/386. Thank you!

closed 2021-02-17 by felipediel