mjg59/python-broadlink #240: How to add devices in domoticz

hexolite, 2019-05-08:
Hi, 
Sorry, I have a noob question. I installed the plugin on Domoticz and I added my RM mini 3 in the hardware page. 
Then I have "broadlink - Remote" and "broadlink - Command" in the devices tab. 

But what is next? How to add my TV or other IR devices to my broadlink and further to Domoticz. 

I'm pretty sure that the answer is simple but I am new to domoticz and to domotic in general so I don't get how to do. 

Thanks in advance,

Whilser, 2019-05-18:
Just read
https://www.domoticz.com/wiki/Plugins/BroadlinkRM2.html

closed 2019-06-01 by Danielhiversen

mzy2240, 2019-06-06:
The latest version (0.11) does not work with the Domoticz out of the box, so my suggestion is to install the 0.10.