mjg59/python-broadlink #252: rm3 mini  don`t work discovery and no error

KubovD, 2019-06-23:
Hi. Sorry for my english. I use domoticz and all be good, before  i need to reinstall and upgrade firmware in rm mini  to 55V.
`Python 3.5.4 (default, Aug 31 2017, 12:01:51)
[GCC 5.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> devices = broadlink.discover(5)
>>> print(devices)
[]
`
But e-control do it well.

Whilser, 2019-06-27:
Try 
pip3 install broadlink==0.10

ricolk, 2019-07-01:
Same issue, also with  
> pip3 install broadlink==0.10

`devices = broadlink.discover(5)`

return me an empty variable.

I have trouble also with SSID configuration. My SSID is not broadcast, but setup() seems not working an I have no feedback from my RM3 mini.
But "discover" doesn't work for me also with an already configured device by app.

gabrieletassoni, 2019-07-01:
I also can't discover my RM 3 mini, setup via ihc app is ok, I can send ir commands to devices from the app, the SSID of my wlan is broadcast, but the raspberry in which I installed the python package, cannot discover, using either the cli or the python library, any device.
The python version I'm using is 2.7

gabrieletassoni, 2019-07-01:
I tested the 

> Try
> pip3 install broadlink==0.10

This works for me, after installing with `sudo pip install broadlink==0.10`, the use of the cli goes right:

`$ ./broadlink_discovery 
Discovering...
###########################################
RM2
# broadlink_cli --type 0x2737 --host 192.168.121.175 --mac 57037542f7c8
Device file data (to be used with --device @filename in broadlink_cli) : 
0x2737 192.168.121.175 57037542f7c8
temperature = 0.0`

AnilDaoud, 2019-07-10:
i suspect same issue as #253 
need to install cryptography

KubovD, 2019-07-10:
> Try
> pip3 install broadlink==0.10

Thanks. Thats work.

closed 2019-07-10 by KubovD