mjg59/python-broadlink #253: must install cryptography, or it fail in silent.

optman, 2019-06-30:
it should not fallback to pyaes, which won't work!

Danielhiversen, 2019-06-30:
Why does not pyaes work?

optman, 2019-06-30:
I don't know,  i found response[0x22]  is not 0, which mean there is an error.  I have try it on different machines, there is only one works. I guess it must be enviroment issue. The one works,  there is no ```pyaes``` installed, but  ```cryptography``` is installed.  And i try to install ```cryptography``` on all other machines, all works. 

BTW, my device is sp2 mini.

AnilDaoud, 2019-07-10:
same issue, i was not able to learn or send any command with only pyaes.
after i installed cryptography, it worked fine
device is rm mini3

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/281. Thank you!

closed 2021-02-17 by felipediel