mjg59/python-broadlink #269: Energy Consumption

tactmaster, 2019-07-09:
Hello,

I am trying to monitor the energy consumption of the SP3S plug socket. Constantly polling the current wattage could work. However I guess that won't be very accurate. The IHC app does measure monthly consumption, is this storage somewhere in the device? 

Thanks

Danielhiversen, 2019-09-20:
Have you tried the `get_energy ` function?

tactmaster, 2019-09-20:
So get_energy() get the instantaneous power computation. However even if you constantly polled  say every second it may not give a accurate reading. As the wattage my vary over the second

Thanks

nihilus, 2020-01-09:
I suppose this might be the difference between sp2 and sp3, right?

rnonymous, 2020-01-29:
So i have a an SP3s (according to Aliexpress)
On the back of it, on the sticker, it says SP3.  In python it recognizes as SP2.

Using the broadlink app I can obtain daily, monthly, and current energy consumption.
I have a feeling that the SP2/3 just sends all this info every second to the chinese mothership. And when we press month view, i get the info stored on the server. 

To confirm this i did a small test:
I unplugged my router. The official app, could still show the current energy consumption, but gave error when trying to obtain monthly/daily.

So yes i think we have to poll the power meter every second and store it. I assume it returns the average power consumption per second, and we don't have to poll it multiple times per second. Anyway, its not meant for that kind of accuracy.

Now my last issue is how do i block it from sending info to the mothership, or having that info sent us instead?

tactmaster, 2020-01-29:
You are probably right...sadly.....may get a induction energy mesaure and double check.

I will check the IP addresses that it uses and get back to you So you can block it. 
They are sold as https://www.clasohlson.com/se/WiFi-Smart-Plug%2C-Clas-Ohlson-Home/p/Pr388192000 so you can use the class oholson home app to configure them. Maybe if you reset them and and use this app it will stop mothership communications

felipediel, 2021-10-18:
Hi. This device does not store sensor data. It keeps pushing the data to the cloud constantly and the sum is computed there. If you calculate the monthly consumption by polling the device, the effect will be the same.

If you don't want to reinvent the wheel, now Home Assistant's Broadlink integration has this feature. It's pretty easy to set up, you can keep things local and you can even analyze the sensor data in grafana. Pretty cool.

closed 2021-10-18 by felipediel