mjg59/python-broadlink #270: broadlink.discover(timeout=5) never comes back in Raspbian Buster

thunder2k, 2019-07-12:
Hi,
I was trying to use this with Raspbian Buster but it doesn't work.

broadlink.discover is stuck when I try to launch it from python console.

Any suggestions?

thunder2k, 2019-07-14:
A small update: now the discover command comes back but apparently no devices are found.

Any ideas?

thunder2k, 2019-07-15:
I've reverted back from 0.11.1 to 0.9 (the one I have on my other environment which is working correctly) and it works flawlessly. So something broke between 0.9 and 0.11.1.

Can you please check it?

Thanks!

Bye

mrlexley, 2019-07-15:
I have the same, tried version 1.11.1 it does work. Reverted back to 0.10 it does work again. Strangely not always getting all my S3 sockets :-(.

thunder2k, 2019-07-15:
I think you meant that you tried 0.11.1 and it DOESN'T work. Then you reverted to 0.10 and it DOES work, is that correct?

mrlexley, 2019-07-15:
Yes, I fixed the post.

mrlexley, 2019-07-15:
Currently I don't have the time to dive into it. But I will check on later if no-one is having time.

letitbe1503, 2019-07-20:
because function socket.gethostbyname(socket.gethostname()) can't get real local ip address 
 
`def discover(timeout=None, local_ip_address=None):
    if local_ip_address is None:
        local_ip_address = socket.gethostbyname(socket.gethostname())
    address = local_ip_address.split('.')`

[image]

in  version 0.9 and 0.10 local_ip_address get from 

`s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.connect(('8.8.8.8', 53))  # connecting to a UDP address doesn't send packets
local_ip_address = s.getsockname()[0]`

if you want function discover work with version 0.10-0.11.1 you can send real local ip address to function discover like this

```
#!/usr/bin/env python

import broadlink
import time
import argparse
import socket

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.connect(('8.8.8.8', 53))  # connecting to a UDP address doesn't send packets
local_ip_address = s.getsockname()[0]
print(local_ip_address)

parser = argparse.ArgumentParser(fromfile_prefix_chars='@');
parser.add_argument("--timeout", type=int, default=5, help="timeout to wait for receiving discovery responses")
# parser.add_argument("--ip", default=None, help="ip address to use in the discovery")
parser.add_argument("--ip", default=local_ip_address, help="ip address to use in the discovery")
args = parser.parse_args()

print "Discovering..."
# devices = broadlink.discover(timeout=args.timeout)
devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip)
for device in devices:
    if device.auth():
        print "###########################################"
        print device.type
        print "# broadlink_cli --type {} --host {} --mac {}".format(hex(device.devtype), device.host[0], ''.join(format(x, '02x') for x in device.mac))
        print "Device file data (to be used with --device @filename in broadlink_cli) : "
        print "{} {} {}".format(hex(device.devtype), device.host[0], ''.join(format(x, '02x') for x in device.mac))
        if hasattr(device, 'check_temperature'):
            print "temperature = {}".format(device.check_temperature())
        print ""
    else:
        print "Error authenticating with device : {}".format(device.host)

```

[image]

thunder2k, 2019-07-20:
Hi,
Is this a change that must be done in this module or in the way this module is invoked from other code?

PR #275 (2019-08-16): Use old IP address lookup logic as fallback
On some machines, resolving the local hostname results in a loopback IP
address (127.0.0.0/8). This breaks discovery. In these situations, fall
back to the old IP address lookup logic that was removed on commit 790edb9.

---

This fixes the regression introduced in PR #244. Discovery was no longer working for me and I was able to bisect the problem to that commit. This should also fix #270.
closed 2019-08-16 by Danielhiversen