mjg59/python-broadlink #273: Broadlink discovery

C-Jay-C, 2019-07-29:
Hi guys, total noob but trying to get this to work on Windows with Pythonwyn.
Basically just want to set up my RM Pro without having to use the Broadlink app and came across this which looks awesome as I can set my RM into access point mode and should be able to use the Broadlink discovery script.

Noticed that I could import the Broadlink library under Linux with the install command but since I'm not running Linux not sure what to do next...
Would anybody be able to assist?

The aim is to just get my device connected to my LAN (already blocked the Mac of the device on my router for any dial-home behaviour) so that I can use it locally in a home automation setup using Hubitat. Someone came up with an API for the Hubitat but it's only useful after the RM Pro has already been set up on the network...

Thx very much in advance!!

PainElemental, 2019-08-09:
Hello,
If you're more a windows guy, you can try Broadlink manager:
https://sourceforge.net/projects/broadlink-manager/

This works for IR-codes, but does not work for RF-codes with newer broadlink devices.

There is also another port of this python library:
https://www.nuget.org/packages/SharpBroadlink/

Haven't tried this yet, but it comes with a command line interface.

Problem with all these ports is, that they get outdated very soon...

Best Regards

felipediel, 2021-02-18:
Hello there.

If you were asking which hub to choose, I would recommend Home Assistant, which depends on this module and offers native support for Broadlink devices.

If you chose Hubitat, that's ok. You can set up an auxiliary HTTP server to work as a bridge between Hubitat and this module. Then you send the HTTP requests via Hubitat.

I know people who achieved this with [broadlinkgo](https://github.com/rob121/broadlinkgo). It is based on this module and comes with a built-in HTTP server. You could also use Home Assistant for this. There are many options and many tutorials on YouTube with step-by-step instructions.

I hope I was helpful. Closing.

closed 2021-02-18 by felipediel