mjg59/python-broadlink #274: RM Mini 3 Broken in 0.11.1

webworxshop, 2019-07-31:
Hi,

As per [this](https://community.home-assistant.io/t/broadlink-rm-mini-3-failed-to-connect-to-device/129081) thread on the Home Assistant forums the RM Mini 3 appears to be broken with the new version of the library (it doesn't get passed the auth stage).

Trying the CLI tools, I am unable to learn or send any commands successfully with version 0.11.1. When I check out version 0.10 however it works fine.

I'm happy to keep investigating further, but if someone could point me in the right direction that would be a big help.

Thanks in advance.

Lisheng2016, 2019-08-01:
still failed to learn packages @0.10 i'm on 508b056.
it always show "No data received"

webworxshop, 2019-08-02:
I am also on 508b056. You environment is probably still referring to another installed version. Here's how I set up my testing environment:

```
git clone https://github.com/mjg59/python-broadlink.git
cd python-broadlink
git checkout 0.10
mkvirtualenv broadlink
pip install -r requirements.txt
python setup.py install
cd cli
```

Then when you run the cli tools you need to prefix them with `python` to make sure that the python interpreter from the virtualenv is called. Without this the shebang will call the system python, which probably has 0.11.1 installed in it's path:

```
python ./broadlink_cli --type 0x2737 --host <ip> --mac <mac> --learn
Learning...
260060000001209313111312131113361312131212371336133614111212131212371311131213361312121213121212131213361312121213361435133613371237131113361337120005690001264914000c3d0001254914000c3d0001264913000d050000000000000000
Base64: JgBgAAABIJMTERMSExETNhMSExISNxM2EzYUERISExISNxMRExITNhMSEhITEhISExITNhMSEhITNhQ1EzYTNxI3ExETNhM3EgAFaQABJkkUAAw9AAElSRQADD0AASZJEwANBQAAAAAAAAAA
```

Lisheng2016, 2019-08-03:
> I am also on [508b056](https://github.com/mjg59/python-broadlink/commit/508b0563eda70a67a04df3986c5ea95723d1c31d). Your environment is probably still referring to another installed version. Here's how I set up my testing environment:
> 
> ```
> git clone https://github.com/mjg59/python-broadlink.git
> cd python-broadlink
> git checkout 0.10
> mkvirtualenv broadlink
> pip install -r requirements.txt
> python setup.py install
> cd cli
> ```
> 
> Then when you run the cli tools you need to prefix them with `python` to make sure that the python interpreter from the virtualenv is called. Without this the shebang will call the system python, which probably has 0.11.1 installed in it's path:
> 
> ```
> python ./broadlink_cli --type 0x2737 --host <ip> --mac <mac> --learn
> Learning...
> 260060000001209313111312131113361312131212371336133614111212131212371311131213361312121213121212131213361312121213361435133613371237131113361337120005690001264914000c3d0001254914000c3d0001264913000d050000000000000000
> Base64: JgBgAAABIJMTERMSExETNhMSExISNxM2EzYUERISExISNxMRExITNhMSEhITEhISExITNhMSEhITNhQ1EzYTNxI3ExETNhM3EgAFaQABJkkUAAw9AAElSRQADD0AASZJEwANBQAAAAAAAAAA
> ```

 it's working after uninstalling system broadlink package.Thanks.

webworxshop, 2019-08-07:
@Lisheng2016 please test my fix from #276 and let me know if this also works for you.

parautenbach, 2019-11-26:
As a workaround, can one manually downgrade to `python-broadlink` v0.10.x or upgrade to v0.12.x?

chemelli74, 2019-11-29:
Hi,

I receive a 56 bytes response to my 0x065 auth packet towards RM3 mini.
As the protocol says to skip first 0x38 bytes, I get an empty array.

Any idea ?

Thx,

Simone

chemelli74, 2019-12-01:
> Hi,
> I receive a 56 bytes response to my 0x065 auth packet towards RM3 mini.
> As the protocol says to skip first 0x38 bytes, I get an empty array.
> Any idea ?
> Thx,
> Simone

Device is identified as "0x2737" (RM mini) but it's a RM mini 3. Shouldn't it be a "0x27c2" ?

Simone

chemelli74, 2019-12-07:
> I am also on 508b056. Your environment is probably still referring to another installed version. Here's how I set up my testing environment:
> git clone https://github.com/mjg59/python-broadlink.git [...]

Unfortunately for me, broadlink_discovery still fails at auth() even with 0.10.
Can you check if it works for you ?
BTW, I had to add () to all print statements.

Simone

paludi, 2019-12-23:
@chemelli74 
Same here, v 0.10 fails for me as well with authentication. After skipping the first 0x38 bytes, I also get an empty array.
I have a pretty new RM mini 3, firmware version v44057.
I did not find time yet to dig deeper, how do I quickly retrieve the device type?

paludi, 2019-12-24:
The hint by @dr-boss in #301 (https://github.com/mjg59/python-broadlink/issues/301#issuecomment-565856599) to use the old broadlink app for initial pairing solved it for me. Discovery and learn mode are working now with v 0.10. Haven't tested more yet. 
Looks like the RM mini 3 uses a different communication mode depending on the initial pairing.

parautenbach, 2019-12-25:
More convoluted, but I had most of this already set up (so it's very specific to what I have/need): I just run the Broadlink platform with Home Bridge and use the Home app on iOS to enable the learn mode. I then capture the hex code and convert it to base64 and add it to my Home Assistant setup – that's when I couldn't find the IR codes for whatever remote I'm learning already on the web.

chemelli74, 2019-12-27:
> More convoluted, but I had most of this already set up (so it's very specific to what I have/need): I just run the Broadlink platform with Home Bridge and use the Home app on iOS to enable the learn [...]

@parautenbach, can you please detail the solution ? Basically which plugin for Homebridge you used and how did you put the device in learn mode trough iOS "home" app. 

Thank you,
Simone

parautenbach, 2019-12-28:
@chemelli74 I'll assume you have or know how to set up Homebridge. You then need to install this plugin: https://www.npmjs.com/package/homebridge-broadlink-rm. 

The docs are very good: https://lprhodes.github.io/slate/. But, basically, just add a very minimal config to your Homebridge config file:

```json
{
  ...
  "platforms": [
      {
        "platform": "BroadlinkRM",
        "name": "Broadlink RM"
      }
  ]
}
```

The learn button will show up automatically on the iOS Home app after restarting Homebridge (assuming you've already added the bridge). 

Tail your Homebridge log file and hit the new learn button and within 10s point a remote to your Broadlink RM device. A hex code will show up:

```
Dec 28 08:14:42 securitypi homebridge[534]: [12/28/2019, 8:14:42 AM] [Broadlink RM] [RESULT] Learn Code (learned hex code: 26007800041b083b072b085b0a270a280829073d070001920a1a094d09160a260b170a580a16091709000a420a19093b0a27085b09290928082a093a0a000190071d072b073c09290916075c0917081809000815091b073c0a28095a0b260a28072a093b090001900d170b220c3c092909160b5809170a160b000d05)
```

Pump this through a hex to base64 converter (or use https://base64.guru/converter/encode/hex):
```sh
$ echo "26007800041b083b072b085b0a270a280829073d070001920a1a094d09160a260b170a580a16091709000a420a19093b0a27085b09290928082a093a0a000190071d072b073c09290916075c0917081809000815091b073c0a28095a0b260a28072a093b090001900d170b220c3c092909160b5809170a160b000d05" | xxd -r -p | base64 -w 0
JgB4AAQbCDsHKwhbCicKKAgpBz0HAAGSChoJTQkWCiYLFwpYChYJFwkACkIKGQk7CicIWwkpCSgIKgk6CgABkAcdBysHPAkpCRYHXAkXCBgJAAgVCRsHPAooCVoLJgooByoJOwkAAZANFwsiDDwJKQkWC1gJFwoWCwANBQ==
```

You can now use this code in your Home Assistant setup.

chemelli74, 2019-12-29:
> @chemelli74 I'll assume you have or know how to set up Homebridge. You then need to install this plugin: https://www.npmjs.com/package/homebridge-broadlink-rm.

Thank you for sharing.
Can you just specify the version of HomeBridge and the broadlink plugin you are using ?

Simone

parautenbach, 2019-12-31:
Homebridge v0.4.50
Broadlink plugin v3.5.5

parautenbach, 2020-03-30:
@felipediel here you go. Do you need any other info?

```
14:13:46,125 root INFO Broadlink API Debug
14:13:46,128 __main__ DEBUG Starting discovery...

14:13:46,145 __main__ DEBUG [Request]
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\xe4\x07\r\x0e\x14\x01\x1e\x03\x00\x00\x00\x00\xc0\xa8\x00h \xbc\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00')

14:13:47,3 __main__ DEBUG [Response]
bytearray(b"\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x00\x00\xe4\x07\r\x0e\x14\x01\x1e\x03\x00\x00\x00\x00\xc0\xa8\x00h \xbc\x00\x00\x04\xd3\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x004\xe0\xf2g7\'j\x00\xa8\xc0\xf8\x01@B\xf7\xc8\xe6\x99\xba\xe8\x83\xbd\xe9\x81\xa5\xe6\x8e\xa7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00")

14:13:51,150 __main__ DEBUG [Device 0]
14:13:51,151 __main__ DEBUG Host: ('192.168.0.106', 80)
14:13:51,152 __main__ DEBUG Mac: f8014042f7c8
14:13:51,153 __main__ DEBUG Type: 0x2737

14:13:51,154 __main__ DEBUG Starting authentication...

14:13:51,155 __main__ DEBUG [Request]
bytearray(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00]\xf7\x00\x007\'e\x00sj\xf8\x01@B\xf7\xc8\x00\x00\x00\x00\xa1\xc3\x00\x00E4R\xe7\xf9.\xda\x95\x83D\x93\x085\xef\x9am\xfbi-\xc3p\xb9\x04C\xac\\\xd6?\xbbS\xad\xfa\x08\x81L\xa7\xf8\xcfAq\x002\x8eW\x0c;\x86\xc9M\x05p\x84I\xa3\x89\xe2\x9a\xe1\x04T6\xa0[\xdd\xdc\x02\xc1a\xaf\x13%\xe8~\x19\xb0\xf7\xd1\xce\x06\x8d\xe5\x1ba\x91V\x87m3\x8c\xff;\x99\x1e@\xcd\xb1")

14:13:51,183 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe9\xd7\x00\x007'\xe9\x03sj\xf8\x01@B\xf7\xc8\x00\x00\x00\x00\x12\xc6\x00\x00\xa6\xa4\xb6\xd6\x9d\xb4'x\x00E\x85\r\x8a\xb4\x9cgn U\xc1\xe5:/s\x06rN`\xc5\xd2\xb0U", ('192.168.0.106', 80))

14:13:51,243 __main__ DEBUG Authentication successful.

14:13:51,244 __main__ DEBUG Entering learning mode...

14:13:51,245 __main__ DEBUG [Request]
bytearray(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00O\xd0\x00\x007\'j\x00tj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb2\xbe\x00\x00q\xaf\x81\xce\x90\x80\xae\x84%Q\x00|B\xc87o")

14:13:51,275 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd6\xd0\x00\x007'\xee\x03tj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb2\xbe\x00\x00q\xaf\x81\xce\x90\x80\xae\x84%Q\x00|B\xc87o", ('192.168.0.106', 80))

14:13:51,277 __main__ DEBUG [Request]
bytearray(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb5\xd0\x00\x007\'j\x00uj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb3\xbe\x00\x00-m$\xe3\xc9\xf2\xe6\xca\xc44!?\xa04\x18g")

14:13:51,303 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00{\xcb\xfb\xff7'\xee\x03uj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xaf\xbe\x00\x00", ('192.168.0.106', 80))

14:13:52,306 __main__ DEBUG [Request]
bytearray(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb6\xd0\x00\x007\'j\x00vj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb3\xbe\x00\x00-m$\xe3\xc9\xf2\xe6\xca\xc44!?\xa04\x18g")

14:13:52,330 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00|\xcb\xfb\xff7'\xee\x03vj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xaf\xbe\x00\x00", ('192.168.0.106', 80))

14:13:53,332 __main__ DEBUG [Request]
bytearray(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb7\xd0\x00\x007\'j\x00wj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb3\xbe\x00\x00-m$\xe3\xc9\xf2\xe6\xca\xc44!?\xa04\x18g")

14:13:53,361 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x89\xf4\x00\x007'\xee\x03wj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\x8e\xc7\x00\x00\x00\xc61'^\xa1\xd2\xd3\x17;<\x11bC 7\x81'\xbc\xd0@\x02\xf7\xfe\x88\x9b\x8bEa\xa8G\xf2\x9d\xa3\x11\xf0@\xaa3\xba-X\xb9\xd3\xb8\x01<\xe2\xfa\xb6\xab)&0\xf2\xd0\x18\x1f\x9a,7\xa798\x9cu`{\x0c\xdd3_\xa7\x81\xf8!\x93GO^pQUHS\x0bj\x88u\xe4\xe7R+q\x92\x9c", ('192.168.0.106', 80))

14:13:53,363 __main__ DEBUG [Code received]
JgBQAAABIZEVEhMSExITExITExITEhM3ExITEhITExITNxM3EzYTExI3EzcTNxM3EzYTExITEzcTEhMTEhMTExITEzcSNxMSEwAFpQABKEgTAA0FAAAAAAAAAAA=

14:13:54,365 __main__ DEBUG Sending code...
14:13:54,367 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00E\x01\x00\x007\'j\x00xj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\x8c\xc7\x00\x00"\xdc\tL7)\'{\xc7\xebO\xc5Ez}3\xa4\tK\xc8\xb4\xfd\x13\xa2?/\xae\xda\xfcY/\x9a\x82\xe0t\xf9\xc5M\x87\xca\xf6<\x05\x15?|\xd0\xd1\xfa\xf8\x13\xc5#(b^.\xad\xff\x98\x11H\xc8l\xfe\x9em\xa1\x8d\xc4\xb5\x9bAf\xe2P\x87\xaew7\xdcZ\xf1\x9a\x90\x85\xbeh\xf9\xec}\xdb@\x02\xdd\x99\x15h\x16\xce\xa1\xb0:\x19\x84\'/\x9em\x03J\xb2')

14:13:54,393 __main__ DEBUG [Response]
(b"Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\xd1\x00\x007'\xee\x03xj\xf8\x01@B\xf7\xc8\x01\x00\x00\x00\xb1\xbe\x00\x00ZB\xf7\x1a\xedG\xc4\xfe16\x96\x05\x8b\xa9ku", ('192.168.0.106', 80))
```

felipediel, 2020-04-19:
@parautenbach You device seems to be authenticating and responding ok.

@webworxshop Devices of type 0x2737 have a known issue. Once connected to the cloud, they block local authentication. You need to [disable the cloud](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499). Please try and let me know if it worked.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/317. Thank you!

closed 2021-02-17 by felipediel