mjg59/python-broadlink #278: understanding and decoding ir codes

mglatz, 2019-08-14:
I noticed that I always get slightly different ir code learned by broadlink on same button press (such as vol up); I would expect that press of same button would produce same codes. 

I guess the difference are caused by insignificant timing errors (or something like this) and Id like to be able somehow to compare them. I have little/no knowledge of how IR protocol works, but I expect the IR command can be converted/plotted on a chart such as 

[image]

from broadlink I get codes such as 

`b'&\x00@\x00\x00\x01#\x89\x104\x11\x11\x12\x11\x11\x11\x112\x12\x12\x10\x13\x0f\x13\x10\x12\x11\x12\x0f\x13\x0f\x12\x114\x10\x12\x105\x0f\x13\x0f\x13\x0f\x13\x105\x0f5\x0f\x13\x114\x10\x12\x0f3\x12\x12\x10\x13\x0f5\x11\x12\x0f\x00\r\x05\x00\x00\x00\x00\x00\x00\x00\x00'
`
or

`b'&\x00@\x00\x00\x01#\x8a\x104\x11\x12\x10\x12\x10\x12\x105\x10\x12\x10\x12\x10\x13\x0f\x13\x0f\x13\x0f\x13\x0f\x13\x0f5\x0f\x13\x0f5\x10\x12\x10\x12\x10\x13\x104\x115\x10\x12\x0f5\x0f\x12\x114\x11\x12\x0f\x12\x105\x11\x11\x11\x00\r\x05\x00\x00\x00\x00\x00\x00\x00\x00'
`

these two codes came from the same button press (lg ac, fan mode). They both work (ac recognizes them) but they are different.
How can I convert those bytes in python to binary, or chart or something meaningful I can compare?

(why am I asking: my lg ac remote has mode button that circulates between 5 different modes. I would like to know if the remote sends command "jump to next mode" or rather sends special code for each mode)

Veda82, 2019-08-25:
Hi @mglatz,

I guess this is what you're looking for:

https://github.com/elupus/irgen

Just use 'raw' as output format.

mglatz, 2019-09-04:
thank you @Veda82 !

closed 2021-10-18 by felipediel