mjg59/python-broadlink #279: Doesn't work after moving to another server.

loglux, 2019-08-24:
I moved Domoticz to another server. Imported a database with devices and settings. 
Copied a BroadlinkRM2 plugin directory to a new server.
Installed Crypto, broadnlink, pyaes. 

When I tried to perform a command the log shows that the command was sent but nothing happens. Don't see any error messages in the log.

thunder2k, 2019-08-26:
It happened the same for me, had to install version 0.9 since following versions had bugs. That one was the last one that worked for me at the time.

loglux, 2019-08-26:
version 0.9 of what? could you provide more detail?

thunder2k, 2019-08-26:
Version 0.9 of this library. 

There's a way to specify the version you want when you pip install it. You have to remove the version you already installed first.

Reading against your initial post made me think that it could be a different issue since I was getting an error.

Anyway it worths a try. Check the version you have installed on your old server if you still have access to it

loglux, 2019-08-26:
Removed broadlink 11 and downgraded to 0.9
pip3 install broadlink==0.9

And it works!
Thanks a lot!

vaiper79, 2019-09-15:
I had the same experience. Had broadlink running flawless for testing on a plain Raspbian (most recent and upgraded) install using an RPi2. Did the same stuff on my octoprint raspberry pi (an RPi3) and ran into problems. The script would tell me the switch was Still Off etc, with nothing at all happening on the sp2 power switch. Looking at tcpdump I could see the broadlink sp2 device "talking" back and forth with the octoprint raspberry. I was trying to switch the sp2 on and off via the script. I was stumped until I saw this post. Downgraded the octoprint broadlink package to 0.9 and was back in business. Thanks!!

felipediel, 2021-02-17:
Closing, since it is not a problem to be solved in our code base.

closed 2021-02-17 by felipediel