mjg59/python-broadlink #282: Broadlink A1 can't find device on network

bj32161, 2019-09-25:
Hi,
I did follow up as your instruction as step below.
1. setup AP mode and using my ubuntu PC connect this device "BroadlinkProv"
2. Install broadlink module "pip install broadlink" 
3. run python (2.7) command line
4. import broadlink
5. devices = broadlink.discover(timeout=5)
Why there is no device list. I did check on my wifi router it show on the list (192.168.1.102) as this screenshot. 
Could you please help? I would like to get realtime data (temp, humidity, light and sound) data.
Thank
[image]

eschava, 2019-09-25:
Please try to increase timeout and use explicit value for local_ip_address parameter for discover method, like
devices = broadlink.discover(timeout=20, local_ip_address='192.168.1.100')

bj32161, 2019-09-25:
Hi Eschava,
I got this.
user@M90n-1:~$ python
Python 2.7.15+ (default, Jul  9 2019, 16:51:35) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> devices = broadlink.discover(timeout=20, local_ip_address='192.168.1.102')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/.local/lib/python2.7/site-packages/broadlink/__init__.py", line 71, in discover
    cs.bind((local_ip_address, 0))
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 99] Cannot assign requested address

eschava, 2019-09-25:
local_ip_address='192.168.1.100', not local_ip_address='192.168.1.102'

bj32161, 2019-09-25:
i also try .100 but result is same.
user@M90n-1:~$ sudo python
Python 2.7.15+ (default, Jul  9 2019, 16:51:35) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import broadlink
>>> devices = broadlink.discover(timeout=20, local_ip_address='192.168.1.100')
>>> 
Not thing show on the list.

felipediel, 2021-03-12:
You forgot to provision the device. Enter AP mode, connect to BroadlinkProv and:
```python3
import broadlink
broadlink.setup('myssid', 'mynetworkpass', 3)
```
Hope it works for A1, we've never tested it before.

closed 2021-03-12 by felipediel