mjg59/python-broadlink #284: Create one socket forever is not the best solution. On some OS there are optimizations which could close/invalidate open socket descriptors

exKAjFASH, 2019-10-01:
The only place where socket "self.cs" is using is located in function send_packet. Device constructor creates socket which exists forever.

I've tested 2 Broadlink smart plugs SP3s and one RM mini 3 in Home Assistant on Raspberry Pi. During this test after restart of HA I'm able use SP3s for 5-20 minutes after that time it is unavailable, call's to auth and update are failed without timeout so next restart of HA fixes this issue - that mean that new socket is created per device and they become available.

I've implemented fix and tested it for few days. Please find it in this PR https://github.com/mjg59/python-broadlink/pull/283/commits/8e24cb3434cb0dfd44f2daa7de25a505b6dfaf4b

Thank you

Taxxian, 2019-10-11:
Hi,
this is really important. Many people (like me) cannot use python-broadlink because every time a broadlink device looses connection to the wifi for a short time, it cannot be controlled by home assistant anymore (broadlink android apps still work fine) until HA is rebooted, which takes time, disrupts  automations and is hard to automate itself.

The pull request was closed because of a complaint about to many opened sockets, I am not a networking pro, but I would rather open to many sockets than not use my HA at all...

Please fix this problem...

exKAjFASH, 2019-10-22:
@Taxxian You can use branch with fix from here https://github.com/exKAjFASH/broadlink_rpi - just put it into custom components folder and use 'broadlink_rpi'

Taxxian, 2019-10-23:
@exKAjFASH 
Thank you, I already copy and pasted all the necessary code in a new PyCharm Project, but it had severe side-effects. I thought I made an error because that was my first contact with Python but your solution produces the exact same side effects:
- Broadlink RM (PRO) sends the same signal for minutes when switching a 433MHz device and never recognizes the device has actually changed its status (did not try IR)
- HA takes at least 10 times as long to boot as before
- Some functions in HA take forever or never get startet at all. For example Restart Home Assistant does not work anymore. Everything that is about updating or installing dockers takes forever.

I tried using the custom component only for the devices that are on the border of wifi range, but that only fixes the RM(PRO) problem, the rest of the side effects occur even when the new component is only used for 1 device.

meetyg, 2019-12-21:
Hi there. I too have been having the known problems with broadlink switch not responding after a while. 
I have found some python code for Home Assistant, that is used for another brand of "smart switch" (although I don't know much about the hardware it uses).
In that code, the socket is recreated for every command, taking care to close the previous instance, if exists.

Please reffer to the `__open_socket` function here:
[Pyswitcher](https://github.com/sagilo/pyswitcherv2/blob/master/pyswitcherv2/switcher.py) 
Do you think that this implementation is still problematic? Maybe this should be the way to go with these problematic devices...

exKAjFASH, 2019-12-24:
Hello. I've installed from scratch Home Assistant 0.103.4 (Hass.io) on Intel NUC i3 device, then added 2 Broadlink plugs and this issue happened again. So I've used my custom component broadlink_rpi to solve it. This mean that it is not related to Raspberry Pi issue and exist on other hardware and builds.

felipediel, 2020-04-25:
@exKAjFASH Thank you!

Fixed by https://github.com/mjg59/python-broadlink/pull/350.

exKAjFASH, 2020-04-27:
Thank you, will switch back to original broadlink integration.

closed 2020-04-27 by exKAjFASH