mjg59/python-broadlink #287: Error on sending commands to RM2 from raspberry, non-hexadecimal number found in fromhex

sanhwy, 2019-10-08:
**sudo ./broadlink_discovery**
Discovering...
###########################################
RM2
# broadlink_cli --type 0x27a9 --host 192.168.1.10 --mac 87554842f7c8
Device file data (to be used with --device @filename in broadlink_cli) :
0x27a9 192.168.1.10 87554842f7c8
temperature = 0.0

**sudo ./broadlink_cli --type 0x27a9 --host 192.168.1.10 --mac 87554842f7c8 --learnfile tv.power**
Learning...
Saving to tv.power

**sudo ./broadlink_cli --type 0x27a9 --host 192.168.1.10 --mac 87554842f7c8 --send tv.power**
Traceback (most recent call last):
  File "./broadlink_cli", line 123, in <module>
    if args.durations else bytearray.fromhex(''.join(args.data))
ValueError: non-hexadecimal number found in fromhex() arg at position 0

**content of file tv.power:**
260058000001269614111411133614111411141114111411143614361312133614361436143614361312131213121336141114111411141114361436133614111436143614361436130005290001274b14000c5e0001274c14000d05

**Where is the problem ?**

sanhwy, 2019-10-08:
this command is working
**sudo ./broadlink_cli --type 0x27a9 --host 192.168.1.10 --mac 87554842f7c8 --send 260058000001269614111411133614111411141114111411143614361312133614361436143614361312131213121336141114111411141114361436133614111436143614361436130005290001274b14000c5e0001274c14000d05**

but when using the file command **--send tv.power** we reveive the error

felipediel, 2021-03-14:
You need to use @ when reading the code from a file, eg:
```
./broadlink_cli --type 0x27a9 --host 192.168.1.10 --mac 87554842f7c8 --send @tv.power
```

closed 2021-03-14 by felipediel