mjg59/python-broadlink #288: pip install fails

vzorglub, 2019-10-09:
pip install broadlink fail:
```
pi@piup:~/broadlink-mqtt $ pip install -r requirements.txt
Collecting paho-mqtt (from -r requirements.txt (line 1))
  Downloading https://files.pythonhosted.org/packages/25/63/db25e62979c2a716a74950c9ed658dce431b5cb01fde29eb6cba9489a904/paho-mqtt-1.4.0.tar.gz (88kB)
    100% |████████████████████████████████| 92kB 690kB/s
Collecting broadlink (from -r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/3f/3f/4d9bdca3c37f7987c9e541edc19db8e93562ba7f967f6de4394a35ba1e1f/broadlink-0.12.0.tar.gz
Collecting PyCRC (from broadlink->-r requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/pycrc/PyCRC-1.21-py2.py3-none-any.whl
Collecting cryptography>=2.1.1 (from broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/c2/95/f43d02315f4ec074219c6e3124a87eba1d2d12196c2767fadfdc07a83884/cryptography-2.7.tar.gz (495kB)
    100% |████████████████████████████████| 501kB 210kB/s
Collecting asn1crypto>=0.21.0 (from cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/8d/c3/be192e56ab85a7b2439caf83049b9936292aaf0398038a2fe28716f18951/asn1crypto-1.0.1-py2.py3-none-any.whl (103kB)
    100% |████████████████████████████████| 112kB 726kB/s
Collecting cffi!=1.11.3,>=1.8 (from cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/93/1a/ab8c62b5838722f29f3daffcc8d4bd61844aa9b5f437341cc890ceee483b/cffi-1.12.3.tar.gz (456kB)
    100% |████████████████████████████████| 460kB 225kB/s
Collecting six>=1.4.1 (from cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting enum34 (from cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://files.pythonhosted.org/packages/fc/d0/7fc3a811e011d4b388be48a0e381db8d990042df54aa4ef4599a31d39853/ipaddress-1.0.22-py2.py3-none-any.whl
Collecting pycparser (from cffi!=1.11.3,>=1.8->cryptography>=2.1.1->broadlink->-r requirements.txt (line 2))
  Downloading https://www.piwheels.org/simple/pycparser/pycparser-2.19-py2.py3-none-any.whl (111kB)
    100% |████████████████████████████████| 112kB 704kB/s
Building wheels for collected packages: paho-mqtt, broadlink, cryptography, cffi
  Running setup.py bdist_wheel for paho-mqtt ... done
  Stored in directory: /home/pi/.cache/pip/wheels/82/e5/de/d90d0f397648a1b58ffeea1b5742ac8c77f71fd43b550fa5a5
  Running setup.py bdist_wheel for broadlink ... done
  Stored in directory: /home/pi/.cache/pip/wheels/a5/a6/2c/80e09b66b76f7528b7294665184fbff2973ad2975b25bc806d
  Running setup.py bdist_wheel for cryptography ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-mGZ446/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpbYKaRlpip-wheel- --python-tag cp27:
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  Package libffi was not found in the pkg-config search path.
  Perhaps you should add the directory containing `libffi.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'libffi' found
  c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
   #include <ffi.h>
                   ^
  compilation terminated.
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-build-mGZ446/cryptography/setup.py", line 275, in <module>
      **keywords_with_side_effects(sys.argv)
    File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
      _setup_distribution = dist = klass(attrs)
    File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 317, in __init__
      self.fetch_build_eggs(attrs['setup_requires'])
    File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 372, in fetch_build_eggs
      replace_conflicting=True,
    File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 846, in resolve
[... 182 lines cut ...]
    copying cffi/backend_ctypes.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/model.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/error.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/_embedding.h -> build/lib.linux-armv7l-2.7/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-2.7/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/c
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python2.7 -c c/_cffi_backend.c -o build/temp.linux-armv7l-2.7/c/_cffi_backend.o
    c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
                     ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-mGZ446/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-uo2PCD-record/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /tmp/pip-build-mGZ446/cffi/
```
On raspberry pi

eschava, 2019-10-09:
Please check if you have libffi-dev package installed
Or execute the command to install it anyway
sudo apt-get install libffi-dev

vzorglub, 2019-10-09:
Thank you
That worked 
I could be worth adding that to the docs
Thanks again

closed 2019-10-09 by vzorglub