mjg59/python-broadlink #289: New device (devtype 0x51e3)

vzorglub, 2019-10-09:
Hi,
I have purchased this device:
https://www.bgelectrical.uk/smart-sockets

After a bit of digging, I found out that they use a broadlink chip:

https://community.home-assistant.io/t/uk-wifi-smart-socket-screwfix-testing/119995/5

So I tried connecting to broadlink-mqtt

Kind of works:
```
$ pi@piup:~/broadlink-mqtt $ python mqtt.py
[2019-10-09 10:27:40,822] DEBUG Connected to 'Unknown' Broadlink device at '192.168.0.99' (MAC c8:f7:42:fc:e5:45) and started listening for commands at MQTT topic having prefix 'broadlink/'
[2019-10-09 10:27:40,834] DEBUG Connected to MQTT broker, subscribing to topic broadlink/#
```

It connects, so there must be a way.
I have a spare device that I can send by post
I can also send debug logs or whatever is needed

also:
```python
>>> import broadlink
>>> print hex(broadlink.discover(timeout=10)[0].devtype)
0x51e3
```
How do I find out what commands it takes?
It could be compatible with an MP1 device but only 2 sockets
How do I add the devtype in the library and then test it locally?

Thanks

vzorglub, 2019-10-09:
Ok, so I have downloaded the code and modified the library
I added the devtype code to the MP1 devices and it connects but the power commands don't work so that theory is gone out the window.

That answered the last question in the post above

There remains the first question: How do I find out what commands the device accepts?

Thanks

vzorglub, 2019-10-13:
Really?? No one is interested in helping to add more hardware to the library?
I am willing to give away the hardware to someone who can.
Thanks

barnybug, 2019-10-13:
I've had a shot at it (see PR above). Working for my test sample of 1. :-)
(Do I get the prize? :+1: )

vzorglub, 2019-10-14:
Not working for me 😢 
Do you need any data?

barnybug, 2019-10-15:
Are the devices returned 'bg1' instances?
If you add some debug that'd help - in the new code after line 363:
`response = self.send_packet(0x6a, packet)`
put:
`print(response.hex())` there
also after line 415:
`payload = self.decrypt(bytes(response[0x38:]))`
put:
`print(payload.hex())` there

Send me the output from a couple of calls.
thanks.

barnybug, 2019-10-15:
Actually judging by your call it's likely the device is returning an error flag in response packet, so it's not accepting our command request. A hard one to debug remotely.

Could you capture some of the actual command packets to the device? I did this on Android by installing "Packet Capture" from the play store, which can sniff the udp packets from the BG Electrical app. Try turning the socket on/off a few times and then look for the last UDP packets going to->from your phone IP to the device IP.

vzorglub, 2019-10-15:
Right
I use python 2
I have added
```python
        key = payload[0x04:0x14]
        print str(binascii.hexlify(key))
```
Nothing prints
I even added a test print at the beginning of the function:
```python
print 'TEST'
```

Nothing prints!!
It looks like the function doesn't run
A search in the file reveals that `auth` is not called from anywhere

```python2
#!/usr/bin/python

import codecs
import json
import random
import socket
import struct
import threading
import time
import binascii
from datetime import datetime

from cryptography.hazmat.backends import default_backend
from cryptography.hazmat.primitives.ciphers import Cipher, algorithms, modes

def gendevice(devtype, host, mac):
    devices = {
        sp1: [0],
        sp2: [0x2711,  # SP2
              0x2719, 0x7919, 0x271a, 0x791a,  # Honeywell SP2
              0x2720,  # SPMini
              0x753e,  # SP3
              0x7D00,  # OEM branded SP3
              0x947a, 0x9479,  # SP3S
              0x2728,  # SPMini2
              0x2733, 0x273e,  # OEM branded SPMini
              0x7530, 0x7546, 0x7918,  # OEM branded SPMini2
              0x7D0D,  # TMall OEM SPMini3
              0x2736  # SPMiniPlus
              ],
        rm: [0x2712,  # RM2
             0x2737,  # RM Mini
             0x273d,  # RM Pro Phicomm
             0x2783,  # RM2 Home Plus
             0x277c,  # RM2 Home Plus GDT
             0x272a,  # RM2 Pro Plus
             0x2787,  # RM2 Pro Plus2
             0x279d,  # RM2 Pro Plus3
             0x27a9,  # RM2 Pro Plus_300
             0x278b,  # RM2 Pro Plus BL
             0x2797,  # RM2 Pro Plus HYC
             0x27a1,  # RM2 Pro Plus R1
             0x27a6,  # RM2 Pro PP
             0x278f,  # RM Mini Shate
             0x27c2  # RM Mini 3
             ],
        a1: [0x2714],  # A1
        mp1: [0x4EB5,  # MP1
              0x4EF7  # Honyar oem mp1
              ],
        hysen: [0x4EAD],  # Hysen controller
        S1C: [0x2722],  # S1 (SmartOne Alarm Kit)
[... 894 lines cut ...]
    # Add the WiFi password to the payload
    pass_start = 100
    pass_length = 0
    for letter in password:
        payload[(pass_start + pass_length)] = ord(letter)
        pass_length += 1

    payload[0x84] = ssid_length  # Character length of SSID
    payload[0x85] = pass_length  # Character length of password
    payload[0x86] = security_mode  # Type of encryption (00 - none, 01 = WEP, 02 = WPA1, 03 = WPA2, 04 = WPA1/2)

    checksum = 0xbeaf
    for i in range(len(payload)):
        checksum += payload[i]
        checksum = checksum & 0xffff

    payload[0x20] = checksum & 0xff  # Checksum 1 position
    payload[0x21] = checksum >> 8  # Checksum 2 position

    sock = socket.socket(socket.AF_INET,  # Internet
                         socket.SOCK_DGRAM)  # UDP
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    sock.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, 1)
    sock.sendto(payload, ('255.255.255.255', 80))
```

barnybug, 2019-10-15:
You need to call 'auth()' yourself.

vzorglub, 2019-10-15:
Ok, so I call `.auth()`
print the key
and send you an hex dump of the packet capture?

vzorglub, 2019-10-15:
Key:
6595f479754ee00b28e8b60ffaf36437

Packet sent to socket:
```
5a a5 aa 55 5a a5 aa 55  00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
ba d9 00 00 e3 51 6a 00  c2 ce 45 e5 fc 42 f7 c8
01 00 00 00 c1 c5 00 00  e3 7c 44 8a 74 80 ad b8
69 36 c6 a8 3c 15 b9 23  42 91 33 a9 4c 1f 01 76
a8 bf 6c 11 3c 78 cd 7d

Packet response from socket
5a a5 aa 55 5a a5 aa 55  00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
e7 01 00 00 e3 51 ee 03  c2 ce 45 e5 fc 42 f7 c8
01 00 00 00 8f e2 00 00  1e eb 2e de 98 52 52 8b
ab 08 73 91 dd 8b c0 18  0b bf 58 ce 7b c0 b9 90
61 31 3b 38 bc a2 7b 23  17 d5 ba aa ea e7 a2 74
24 79 21 92 f0 34 34 c3  b7 a7 d5 b3 f2 04 12 9e
ca 18 ed 45 e4 c5 1f 98  2f 56 7b 62 1f 12 14 b3
46 4a 8a 45 73 62 5d 92  3c 00 df 85 98 70 ae 75
30 02 f5 40 11 e6 13 f9  12 51 a8 49 e8 3a e9 b8
3a ab c7 1d fc b2 3f 85

```

vzorglub, 2019-10-15:
Ok, after doing `.auth()` then it works!!
Apologies

Good work!! You are a start.

barnybug, 2019-10-16:
Great, thanks!

PR #290 (2020-01-30): Add support for BG Electrical Smart Sockets
This adds support for these sockets which basically send json over the broadlink protocol:
https://www.bgelectrical.uk/smart-sockets

Fixes #289
closed 2020-01-30 by mjg59