mjg59/python-broadlink #294: Crypto errors when running on lubuntu 18.04

ajuniper, 2019-11-03:
Am seeing this error whenever my Broadlink device responds:
```
$ ./broadlink_discovery
Discovering...
Traceback (most recent call last):
  File "./cli/broadlink_discovery", line 13, in <module>
    devices = broadlink.discover(timeout=args.timeout, local_ip_address=args.ip)
  File "/usr/local/lib/python3.6/dist-packages/broadlink/__init__.py", line 137, in discover
    dev = gendevice(devtype, host, mac)
  File "/usr/local/lib/python3.6/dist-packages/broadlink/__init__.py", line 58, in gendevice
    return device_class(host=host, mac=mac, devtype=devtype)
  File "/usr/local/lib/python3.6/dist-packages/broadlink/__init__.py", line 365, in __init__
    device.__init__(self, host, mac, devtype)
  File "/usr/local/lib/python3.6/dist-packages/broadlink/__init__.py", line 162, in __init__
    self.update_aes(key)
  File "/usr/local/lib/python3.6/dist-packages/broadlink/__init__.py", line 165, in update_aes
    self.aes = Cipher(algorithms.AES(key), modes.CBC(self.iv),
  File "/usr/lib/python3/dist-packages/cryptography/hazmat/primitives/ciphers/modes.py", line 92, in __init__
    raise TypeError("initialization_vector must be bytes")
TypeError: initialization_vector must be bytes
```

hnuestedt, 2020-05-06:
I am not sure if there are any impacts on different systems or the functionality, but I changed line 185 from
`self.aes = Cipher(algorithms.AES(key), modes.CBC(self.iv),`
to
`self.aes = Cipher(algorithms.AES(key), modes.CBC(bytes(self.iv)),`
and now my device is at least discoverable.

alexkutsan, 2020-05-21:
I have the same issue

Ircama, 2020-06-06:
I had this issue and did a reinstall/upgrade to fix it:

```bash
pip install --upgrade --force-reinstall git+https://github.com/mjg59/python-broadlink.git
```

Check *cryptography* version:

```bash
pip freeze | grep cryptography
```

Mine (which works) is
```
cryptography==2.9.2
```

I see that it is currently set to [2.6.1](https://github.com/mjg59/python-broadlink/blob/ca8a5c6a8b202713eaf97abf36dffba6a65810f5/requirements.txt#L1) in *requirements.txt*.

felipediel, 2021-02-17:
Fixed with https://github.com/mjg59/python-broadlink/pull/505. Thank you!

closed 2021-02-17 by felipediel