mjg59/python-broadlink #301: "Broadlink RM4c mini" support

ArchGalileu, 2019-11-20:
Hello

I bought "Broadlink RM4c mini" but it doesn't work with python-broadlink

It see it but don't connect.

on broadlink_discovery:
`Discovering...
Error authenticating with device : ('192.168.1.6', 80)`
on broadlink_cli:
`Traceback (most recent call last):
  File "./broadlink_cli", line 95, in <module>
    mac = bytearray.fromhex(values[2])
ValueError: non-hexadecimal number found in fromhex() arg at position 2`

Best regards
ArchGalileu

skrutt, 2019-11-21:
What python version are you using?

ArchGalileu, 2019-11-22:
Python 3.8.0

jondavidsson, 2019-12-07:
Stupid me also got the new RM4 mini and was planning to use it with home-assistant. Unfortunately  I didn't check if it was compatible and it seems to not be. 

First of all. I am a total newbie to python and wouldn't consider me as a coder. So everything I do is by trial and error.

This is what I have found out.

I believe the device ID for my RM4 is: 0x610f

Got it from:
hex(devices[0].devtype)
'0x610f'

The problem I seem to have is with the devices[0].auth():
devices[0].auth()
False

I tried add the device ID and hoped that it would be pretty much the same as the other RM devices. 
devices[0].get_type()
'RM2'

But probably since auth() give me false it doesn't matter.

Anyone who can help me point me where to go next?

skrutt, 2019-12-07:
Auth only works the first time, after that a new key is used for the encoding so it is not recognised. 

But you seem to get false even om the first call, so i would print out the sent/received data and see if it matches the expected values.

You can also use wireshark and analyse the traffic, see if there are differences between what you are sending and for example what a compatible app is sending.

skrutt, 2019-12-07:
Maybe you can also have a look at #255

jondavidsson, 2019-12-07:
Thanks for the response!

> Auth only works the first time, after that a new key is used for the encoding so it is not recognised.

With first time, do you mean the first time I connect the device to the network? 
If so, since I’ve already connected and configured it in the Broadlink app auth() was supposed to give me false? Might have to try to reset it and start over.  Also have to try it from my Mac so I can use Wireshark, since I only tried it through ssh on a Linux server. 

> Maybe you can also have a look at #255

Thanks for the referral. I’ll look into it :)

Ha en fortsatt trevlig helg 😁

skrutt, 2019-12-07:
I was a bit unclear, auth only works once after the device object is created, so if i do:

Discovery
Auth -> true
Auth -> false

But i would Guess that this isnt really your problem.. but a good thing to Keep in mind

chemelli74, 2019-12-08:
> Stupid me also got the new RM4 mini and was planning to use it with home-assistant. Unfortunately  I didn't check if it was compatible and it seems to not be.
> First of all. I am a total newbie to [...]

Hi, same result here with a "RM mini 3" (firmware v44057):

  import broadlink
  devices = broadlink.discover(timeout=5)
  hex(devices[0].devtype)
**'0x2737'**
  devices[0].type
**'RM2'**
  devices[0].auth()
**False**

Response to auth request is a 56bytes packet, that means a empty auth() key.
See also issue #274 

Simone

jondavidsson, 2019-12-08:
I also get a 56 byte response on the RM4 according to Wireshark.
This is the packets from auth()
`0000   60 03 08 89 5e ce 24 df a7 34 bf 4f 08 00 45 00`
`0010   00 54 05 c3 00 00 ff 11 9f a9 0a 00 01 24 0a 00`
`0020   01 09 00 50 f6 6c 00 40 0b 2a 5a a5 aa 55 5a a5`
`0030   aa 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00`
`0040   00 00 00 00 00 00 00 00 00 00 8d ca ff ff 2a 27`
`0050   e9 03 29 25 4f bf 34 a7 df 24 00 00 00 00 af be`
`0060   00 00`

> Maybe you can also have a look at #255

I changed the if not payload from False to True according to above pull request and added the device ID and after that the broadcast cli could discover it. But I can't learn anything from the cli.

Hopefully someone with the knowledge can take a look at this since my knowledge is not near enough to know what to do.

chemelli74, 2019-12-08:
> I changed the if not payload from False to True according to above pull request and added the device ID and after that the broadcast cli could discover it. But I can't learn anything from the cli.

For the discovery to work, it's enough to add the device ID.
The auth() is not as easy as forcing a "true" state, unfortunately.

Simone

jaruba, 2019-12-09:
@jondavidsson 

> I believe the device ID for my RM4 is: 0x610f
> Got it from:
> hex(devices[0].devtype)
> '0x610f'

That's strange.. my RM4 Mini devtype is `24846` which translates to `0x610e`, not `0x610f`..

jondavidsson, 2019-12-09:
From this BroadLink homebridge thread there’s the same issue. One device seems to have the same device is as me. 

If what user dereklucas says about the device is only controllable through the cloud API then I also suspect new Broadlink (or BestCon which seems to be the new name the use) products will not be compatible with this plugin in the current state. 

@jaruba I bought mine from China, can that be a difference between our devices?

jaruba, 2019-12-09:
It's very possible, Broadlink seems to make region specific devices now, I bought mine from Romania and it didn't want to let me add it to the Broadlink app due to regional restrictions, changing my account to USA also didn't help, changing it again to Germany allowed me to add the device.

I also saw that aliexpress sells a "RM4 Mini NON-EU", so the difference between our devices might be a region specific product release.

jaruba, 2019-12-09:
@jondavidsson Just to clarify, you have the RM4 Mini (as me), right? Not the RM4C Mini... which is a different product.

jondavidsson, 2019-12-09:
> @jondavidsson Just to clarify, you have the RM4 Mini (as me), right? Not the RM4C Mini... which is a different product.

@jaruba Didn’t realize there was two different rm4 mini. Mine is the RM4c. So the device ID difference make sense then. But as you say, there could be a regional difference as well.

jondavidsson, 2019-12-09:
I’ll place an order of an RM mini 3 from Sweden and cross my fingers it’ll work since there seems to be some who also have the same issue with that one.

jaruba, 2019-12-09:
The difference between RM4C Mini and RM4 Mini is that RM4 Mini also supports changing the power cable to a sensor capable one, while RM4C Mini has no such support for additional sensors.

For example, I bought the RM4 Mini and a separate power cord for it that has temperature and humidity sensors on it. This is the only difference between these 2 IR blasters that I know of.

jaruba, 2019-12-09:
RM mini 3 will most probably work without issues, but I was excited about this new IR blaster with modular sensor support.. :(

jondavidsson, 2019-12-09:
Yeah I saw that cable when browsing for a RM mini 3. Luckily they where out of stock 😂

dr-boss, 2019-12-15:
The same problem (error:Traceback (most recent call last): File "./broadlink_cli", line 95, in <module> mac =...)_  occurs after last fw update on RM3PRO if it added to broadlink android app,they not work after this with python-broadlink after first auth. For me at the moment work this method:

1. remove RM3PRO from new app, 
2. reset it manually
3. add by old e-control to network , remove and reset if you like
4 they return to discovered and setup by python-broadlink and work without problems.

I haven't RM4... for test in this moment.

jaruba, 2019-12-16:
@dr-boss The IHC app doesn't allow adding RM4 Mini devices at all, only the Broadlink app works with them.

It's seems clear that Broadlink is moving everything to Broadlink Cloud. Their Private Cloud API is pretty messy, I managed to get the temperature / humidity data from the RM4 Mini sensor, but all the codes regarding the request are hardcoded for now, so it's not something that I can share..

I got the request data by forcing the Broadlink app through a proxy..

chemelli74, 2019-12-17:
> It's seems clear that Broadlink is moving everything to Broadlink Cloud. Their Private Cloud API is pretty messy, 

There is a SDK that you can request if you are building an application. But dunno if you can then share the resulting code with a opensource code.
I bet the only thing to do is a cloud service that has open API on one side to get request from users and Broadlink SDK on the other hand to call Broadlink Cloud.

Simone

paludi, 2019-12-24:
> The same problem (error:Traceback (most recent call last): File "./broadlink_cli", line 95, in mac =...)_ occurs after last fw update on RM3PRO if it added to broadlink android app,they not work after this with python-broadlink after first auth. For me at the moment work this method:
> 
> 1. remove RM3PRO from new app,
> 2. reset it manually
> 3. add by old e-control to network , remove and reset if you like
>    4 they return to discovered and setup by python-broadlink and work without problems.
> 
> I haven't RM4... for test in this moment.

Also works for the RM mini 3

antag0n1st, 2020-03-03:
Will this device ever be supported ?

felipediel, 2020-03-05:
Hi guys. Maybe I can help you. But I don't have this device, so it's up to you to collect the debug data.

1. Let's start with a simple debug.
1.1. Run [this script](https://drive.google.com/open?id=1u0y6onVgIgOQ_u4PL404vzIwjMam0rDy)
1.2. If the light turns on, capture a code
1.3. Save the file debug.txt somewhere else

2. Now we are gonna try to authenticate the device locally
2.1. Remove the device from the official app
2.2. Reset the device
2.3. Add the device to your local network
2.4. Do not set up a cloud (IMPORTANT!)
2.5. Repeat the debug from step 1 (make sure to save debug.txt first or it will be overwritten)
2.6. Send me the files

Critical-Impact, 2020-03-05:
I've got a Broadlink RM Mini 3, FCC ID is 2ACDZ-MMINI3-RM
Now I know this ticket is about the RM4c mini but I assume it's going to be a similar issue. I've run your script and get
Traceback (most recent call last):
  File "debug.py", line 1026, in <module>
    device.send_data(code)
  File "debug.py", line 610, in send_data
    packet += data
TypeError: can't concat NoneType to bytearray

I have dumped the debug.txt and put it to gist. I am only able to connect the module to the new mobile app and not the old app. I can't really add the device after resetting unless I use the cloud app.

https://gist.github.com/Critical-Impact/9386a2c5c43f7311d60278bcabbaae6b

felipediel, 2020-03-05:
@Critical-Impact Your device is a Broadlink RM 3 Mini 0x5f36 v44057. Your problem is already solved and waiting to be merged.

The RM4c device has type 0x610f, but there may be variations. If someone has this device and is looking for a solution, I’m willing to help, but as I said I don’t have the device and I will need some debug data to understand what’s going on.

Critical-Impact, 2020-03-05:
Apologies, I did not realise. If you need any testers for the RM3 Mini just send me a message otherwise just ignore what I've posted here :)

antag0n1st, 2020-03-06:
@felipediel , How do I run the script ? Where should I put the script ? Which lights do I expect to see turn on ?
Sorry I'm very new to this , I need some guidance. I have the RM4c mini , I will provide all the debug info that is needed.

felipediel, 2020-03-06:
You must have Python 3 installed. Then you download the script, open the folder in a terminal and type:
```
python3 debug.py
```
Then:
- If the light turns on you try to capture a command
- Put the file "debug.txt" in your cloud (e.g. Google Drive) and give me the address

bblacey, 2020-03-06:
@felipediel - thanks for offering to help, albeit without a device.  I have several RM Pros on my network that I left in place when running 1.1.1 with the RM4 Mini on my network.  The RM Pro LEDs turned on but the RM4 Mini LED did not.  Here are the results:

**Console Output**
```
$ python /Users/blacey/Downloads/debug.py
Traceback (most recent call last):
  File "/Users/blacey/Downloads/debug.py", line 1022, in <module>
    device.enter_learning()
AttributeError: 'device' object has no attribute 'enter_learning'
```

The debug.txt file contains information about successful access to two RM Pros that I will leave out here to keep the post short but happy to provide if needed.  The relevant RM4 **debug.txt excerpt** follows:

```
07:39:03,262 __main__ DEBUG [Device 2]
07:39:03,262 __main__ DEBUG Host: ('192.168.0.82', 80)
07:39:03,262 __main__ DEBUG Mac: 2ead34a7df24
07:39:03,262 __main__ DEBUG Type: 0x51da

07:39:03,262 __main__ DEBUG Starting authentication...

07:39:03,262 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb6\xf7\x00\x00\xdaQe\x00pz.\xad4\xa7\xdf$\x00\x00\x00\x00\xa1\xc3\x00\x00E4R\xe7\xf9.\xda\x95\x83D\x93\x085\xef\x9am\xfbi-\xc3p\xb9\x04C\xac\\\xd6?\xbbS\xad\xfa\x08\x81L\xa7\xf8\xcfAq\x002\x8eW\x0c;\x86\xc9M\x05p\x84I\xa3\x89\xe2\x9a\xe1\x04T6\xa0[\xdd\xdc\x02\xc1a\xaf\x13%\xe8~\x19\xb0\xf7\xd1\xce\x06\x8d\xe5\x1ba\x91V\x87m3\x8c\xff;\x99\x1e@\xcd\xb1')

07:39:03,278 __main__ DEBUG [Response]
(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd0\xcb\xff\xff\xdaQ\xe9\x03pz.\xad4\xa7\xdf$\x00\x00\x00\x00\xaf\xbe\x00\x00', ('192.168.0.82', 80))

07:39:03,278 __main__ DEBUG Authentication error.

07:39:03,278 __main__ DEBUG Entering learning mode...
```

Let me know what else I can do to provide the information that you need.

felipediel, 2020-03-06:
@bblacey There was an authentication error. Can you control this device with the official app even if you are not connected to the Internet?

Now that you know your device type, you can debug like this:
```
python3 debug.py 0x51da
```

Let's try something that worked for other device.
- Do [this workaround](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499).
- [Debug](https://drive.google.com/open?id=1Gr-Qe_sdVL-Yn2g2ITDFYB1TTpd1pX8Z) again, try to learn and send me the file debug.txt (I need all the file).

bblacey, 2020-03-07:
@felipediel, ok I have tried all of the above including testing a brand new RM4 Mini.  In all cases, I didn't do anything in the Broadlink app except add the device to my local network (stopped as soon as it was added).  I'm using a python 3.8.2 within a conda environment (isolated).  I'm attaching the debug.txt file in case it is useful (two RM Pros, one Mini RM3 and one Mini RM4).  Let me know your thoughts/ideas on how to proceed.
[debug.txt](https://github.com/mjg59/python-broadlink/files/4302510/debug.txt)

FYI, I noticed that your most recent `debug.py` requires argument #1, you might want to make it optional.

felipediel, 2020-03-07:
@bblacey [Here](https://drive.google.com/open?id=1yM6dbnoqWdgNeyfN_-rwrFHXXnzMIck3) is the communication with your device. Make sure to disable line wrap.

Do you see the last byte of the first packet? I've seen this byte before. This byte is evil. It means your device must be connected to the cloud to communicate. And worst, this byte is permanent. Once activated, you need to reset the device to disable it.

The first step in solving the problem is to configure your device locally so that this byte is disabled.  Don't ask me how to do this. You should try things that have already worked for other devices. We disabled this byte on 0x5f36 with the solution I proposed above. If it doesn't work for your device, you need to find another way.

Remember, you need to get rid of the cloud. You can try things like setting up this device using other applcations (e-Control, IHC, Broadlink Manager, etc) or configuring it in an environment without Internet access.

I will only be able to help you after you disable this byte.

Ah thanks for letting me know that the debug script was not working properly. It is working now.

bblacey, 2020-03-08:
@felipediel, Thanks.  Are you talking about the `0xb1` byte at the end of the packet the Auth command (0x0065)?  If so, I see that byte on one of my RM Pros that is working...  Perhaps I'm looking at the "first" packet incorrectly.

That said, here is what I just tried.

• Configured my DHCP server to vend a static IP address to the test RM4 Mini by Mac address
• Configured my firewall to block all traffic from the RM4 Mini static IP address to the Internet (confirmed blocking with logging)
• Reset RM4 Mini with short press
• Put RM4 Mini into AP mode with long press
• Connected to RM4 Mini AP Wifi from Mac
• Wrote and ran a short python program to use the `broadlink.setup()` method to send the SSID and PW to the RM4 Mini.  RM4 Mini rebooted and connected to Wifi SSID using DHCP static address assigned.
• Ran latest `debug.py`.  When the RM4 Mini LED blinked indicating learn mode, I sent several IR codes to the device.

Here are the results [debug.txt](https://github.com/mjg59/python-broadlink/files/4302795/debug.txt).  `debug.py` collected "more" but still I'm still seeing authentication issues.  Let me know if I am on the right track or not.

Also, [debug.txt](https://github.com/mjg59/python-broadlink/files/4302795/debug.txt) includes all my RM devices because if I pass 0x51da on the command-line, `debug.py` fails to discover the RM4 Mini.

Thanks for all your help.

felipediel, 2020-03-08:
@bblacey No, the last byte of the first packet in [the file](https://drive.google.com/open?id=1yM6dbnoqWdgNeyfN_-rwrFHXXnzMIck3) I sent you. The command is a HELLO_RESPONSE (0x7).

If you look at the debug file, this responses are on top (Starting discovery...)
Check the last byte: Zero for all devices you can control locally. Evil byte set in the last response. That came from your 0x51da.

So now your mission is to disable that byte. Remember, you have to find a way to connect this device to your local network without connecting it to the cloud. If the cloud is set (evil byte), you must remove the device from apps, reset it and start all over again until you succeed.

felipediel, 2020-03-08:
@bblacey [Here](https://drive.google.com/open?id=1PwTMcFEw2f8im9m-R0RX9Q2qRcp6JZmd) is one of the last debugs you sent me. The device failed to authenticate and did not enter learning mode (check error codes). Did you mention the led blinked during debug? We need to investigate this, because according to the debug data this could not have happened

bblacey, 2020-03-08:
@felipediel - Well, we have a little more progress with the RM4.

In this test, I pulled the Internet cable from the back of my router to be absolutely certain that a brand-new RM4 Mini could not connect to the Internet.  I powered up a brand-new, never-used RM4, put it in AP mode and used my small python program to join my WiFi (i.e. no Broadlink software touched this RM4).  Once the RM4 joined my WiFi, I immediately ran your `debug.py` and it authenticated with the RM4!  That proves that the evil byte is not set at the factory, but set some other way (like you said, most likely through access to the cloud).  Here is the [debug.txt](https://github.com/mjg59/python-broadlink/files/4303205/debug.txt) output in case you want to take a look.

Also, during this test, the RM4 Mini LED did NOT blink - I'm thinking the two slow blinks that I saw during my earlier test might have indicated that the authentication failed or that the RM4 was talking to the cloud.

Finally, in my earlier tests, I blocked outbound IPv4 & IPv6 TCP/UDP traffic for the RM4 so I am not sure why I couldn't clear the evil byte once it was set.  I'll try resetting the one with the evil byte set while the Internet is completely disconnected to see if I can recover it.

I have installed your hacking tools but haven't had a chance to play with them yet because I wanted to send this interim progress to you in case you have other ideas to try (I'm new to all this).

tobiasf88, 2020-03-08:
Hello everyone,
maybe my debug.txt can help. (Set it up like the workaround and then "python debug.py from @felipediel 
[debug.txt](https://github.com/mjg59/python-broadlink/files/4303226/debug.txt)

bblacey, 2020-03-08:
@tobiasf88, cool.

Was this a new RM4 never touched by the Broadlink software?  I have only been able to authenticate with a new RM4 that has never been touched by any Broadlink software.  In all cases, mine are RM4 Mini Univ "b".  

Given the different device ID, is yours an RM4 Mini Univ "c"?

I'm asking because I just tried to reset two different RM4b devices that were previously paired with the Broadlink app and it seems that I can not get them to authenticate.  I'm thinking that pairing changes some nvram settings that a factory reset doesn't clear.

bblacey, 2020-03-08:
@felipediel, two small updates:

**Summary:**  I am able reset and authenticate an RM4b Mini but can not learn codes (yet).  Resolved filtering bug in `debug.py`

**Details:**
1) Users can remove the evil HELLO byte in a previously paired RM4b Mini by "deleting" the device within the Broadlink App while it has access to the Internet.  If done correctly, the RM4b Mini LED will begin flashing rapidly indicating factory mode.  This removes the evil 0x01 trailing byte in the HELLO response payload.
2) If found out why the `debug.py` device type filtering does not work in the current version.  You need to change `device` to `device.devtype` in the logical equivalence test.  It should be `if dev_type is not None and device.devtype != dev_type:`

felipediel, 2020-03-08:
@bblacey Great, this is huge progress. The authentication step is solved. This device can operate locally. Now let me make my [first attempt](https://drive.google.com/open?id=1hBJNJfaRxfMo9rDgZZCRmpJy0UuIHDMO). Maybe this device waiting for a special header that I've seen before. Do it like debug.

bblacey, 2020-03-08:
@felipediel - great job on your first attempt!  [test_solution1.txt](https://github.com/mjg59/python-broadlink/files/4303508/test_solution1.txt) attached for your examination/inspection.

felipediel, 2020-03-08:
@bblacey You won't believe. Your problem is solved hahaha.
I will make a pull request tonight. Thank you for collaborating to the solution.

bblacey, 2020-03-08:
@felipediel - likewise!

felipediel, 2020-03-08:
@bblacey I will do it. Thank you!

bblacey, 2020-03-08:
@felipediel, just for fun, I fetched your pull request #317 locally, checked out the branch, added `0x51da` to `__init__.py`, ran `pip install -e .` from the pull request branch and voila, I was able to use `broadlink_cli --learn` to record a command from an RM4b!  Looks good to me! 👍

felipediel, 2020-03-08:
Whoever has this same device with a different type, please send me the type so that we can test it and include it in the PR.

tobiasf88, 2020-03-08:
> @tobiasf88, cool.
>
> Was this a new RM4 never touched by the Broadlink software? I have only been able to authenticate with a new RM4 that has never been touched by any Broadlink software. In all [...]

No, the RM4C was touched by the Broadlink App from IOs and IHC europe. Then i reset the Device and add it to the network over AP-Mode. And it look like auth work. ( I kill the app as soon as network connectivity was established.)

tobiasf88, 2020-03-08:
@felipediel  here is my 
[test_solution1.txt](https://github.com/mjg59/python-broadlink/files/4303785/test_solution1.txt)
 from your **test_solution1.py**. It looks like not everything works for the rm4c (**0x610f**) yet. Can you do anything with it? Thanks for your effort.

felipediel, 2020-03-08:
@tobiasf88 The file was not yet updated with your device type. Please [try again](https://drive.google.com/open?id=1hBJNJfaRxfMo9rDgZZCRmpJy0UuIHDMO).

tobiasf88, 2020-03-08:
@felipediel [test_solution1.txt](https://github.com/mjg59/python-broadlink/files/4303811/test_solution1.txt) it works!!!! great man, nice work!!!

bblacey, 2020-03-08:
> @bblacey Just to keep things organized... what is your firmware?

Good question because I don't dare access the RM4b with the Broadlink App ;)  I looked at the `broadlink_cli` to see if it supports firmware interrogation but apparently not.  Is there a command or utility other than the Broadlink app that I can use to get the firmware version?

felipediel, 2020-03-08:
@bblacey Nah, don't worry about that, it's just protocol. You better keep your device safe from that cloud. Peace.

shuenpin, 2020-03-12:
@felipediel 
I also bought the RM4c.
When I executed the debug.py,the devices type show "0x62be" and "This is device does not support enter learning".

```
03:36:03,441 root INFO Broadlink API Debug
03:36:03,441 __main__ DEBUG Starting discovery...

03:36:03,447 __main__ DEBUG [Request]
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\x00\xe4\x07$\x03\x14\x05\r\x03\x00\x00\x00\x00\xc0\xa8X\x0c\xa5\xdc\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00')

03:36:03,462 __main__ DEBUG [Response]
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\x00\xe4\x07$\x03\x04\x06\x0e\x03\x00\x00\x00\x00\xc0\xa8X\x0c\xa5\xdc\x00\x00\x07\xd4\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9b\xed\xd6\x0e\xbeb\x13X\xa8\xc0d\x86O\xa7\xdf$\xe6\x99\xba\xe8\x83\xbd\xe9\x81\xa5\xe6\x8e\xa7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00')

03:36:08,453 __main__ DEBUG [Device 0]
03:36:08,454 __main__ DEBUG Host: ('192.168.88.19', 80)
03:36:08,456 __main__ DEBUG Mac: 64864fa7df24
03:36:08,456 __main__ DEBUG Type: 0x62be

03:36:08,457 __main__ DEBUG Starting authentication...

03:36:08,458 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00P\xf8\x00\x00\xbebe\x00\xbd\xa8d\x86O\xa7\xdf$\x00\x00\x00\x00\xa1\xc3\x00\x00E4R\xe7\xf9.\xda\x95\x83D\x93\x085\xef\x9am\xfbi-\xc3p\xb9\x04C\xac\\\xd6?\xbbS\xad\xfa\x08\x81L\xa7\xf8\xcfAq\x002\x8eW\x0c;\x86\xc9M\x05p\x84I\xa3\x89\xe2\x9a\xe1\x04T6\xa0[\xdd\xdc\x02\xc1a\xaf\x13%\xe8~\x19\xb0\xf7\xd1\xce\x06\x8d\xe5\x1ba\x91V\x87m3\x8c\xff;\x99\x1e@\xcd\xb1')

03:36:08,480 __main__ DEBUG [Response]
(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1a\xdb\x00\x00\xbeb\xe9\x03\xbd\xa8d\x86O\xa7\xdf$\x00\x00\x00\x00N\xc8\x00\x00\xa4\xf0*\x97\xa51q\xec\xa4H`\xa0\xeb\x81\xa4\x15\x91^\x1b\x9ca\xce\xd9y)o\x80\x9d\xa6\xfc\xd8\x16', ('192.168.88.19', 80))

03:36:08,481 __main__ DEBUG Authentication successful.

03:36:08,482 __main__ DEBUG Entering learning mode...

03:36:08,482 __main__ DEBUG This is device does not support enter learning.
```

I try add the device type "0x62be" to debug.py.
The result show "Failed to learn: no code received.".

```
03:42:06,266 root INFO Broadlink API Debug
03:42:06,267 __main__ DEBUG Starting discovery...

03:42:06,272 __main__ DEBUG [Request]
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\x00\xe4\x07*\x03\x14\x05\r\x03\x00\x00\x00\x00\xc0\xa8X\x0c\xa6\xc2\x00\x00\x00\x00\x00\x00\x00\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00')

03:42:06,287 __main__ DEBUG [Response]
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\x00\xe4\x07*\x03\x04\x06\x0e\x03\x00\x00\x00\x00\xc0\xa8X\x0c\xa6\xc2\x00\x00\xf4\xd3\x00\x00\x00\x00\x07\x00\x00\x00\x00\x00\x00\x00\x00\x00\x9b\xed\xd6\x0e\xbeb\x13X\xa8\xc0d\x86O\xa7\xdf$\xe6\x99\xba\xe8\x83\xbd\xe9\x81\xa5\xe6\x8e\xa7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00')

03:42:11,278 __main__ DEBUG [Device 0]
03:42:11,279 __main__ DEBUG Host: ('192.168.88.19', 80)
03:42:11,280 __main__ DEBUG Mac: 64864fa7df24
03:42:11,280 __main__ DEBUG Type: 0x62be

03:42:11,280 __main__ DEBUG Starting authentication...

03:42:11,281 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00Z\xf8\x00\x00\xbebe\x00\xc2\xadd\x86O\xa7\xdf$\x00\x00\x00\x00\xa1\xc3\x00\x00E4R\xe7\xf9.\xda\x95\x83D\x93\x085\xef\x9am\xfbi-\xc3p\xb9\x04C\xac\\\xd6?\xbbS\xad\xfa\x08\x81L\xa7\xf8\xcfAq\x002\x8eW\x0c;\x86\xc9M\x05p\x84I\xa3\x89\xe2\x9a\xe1\x04T6\xa0[\xdd\xdc\x02\xc1a\xaf\x13%\xe8~\x19\xb0\xf7\xd1\xce\x06\x8d\xe5\x1ba\x91V\x87m3\x8c\xff;\x99\x1e@\xcd\xb1')

03:42:11,304 __main__ DEBUG [Response]
(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00$\xdb\x00\x00\xbeb\xe9\x03\xc2\xadd\x86O\xa7\xdf$\x00\x00\x00\x00N\xc8\x00\x00\xa4\xf0*\x97\xa51q\xec\xa4H`\xa0\xeb\x81\xa4\x15\x91^\x1b\x9ca\xce\xd9y)o\x80\x9d\xa6\xfc\xd8\x16', ('192.168.88.19', 80))

03:42:11,306 __main__ DEBUG Authentication successful.

03:42:11,306 __main__ DEBUG Entering learning mode...

03:42:11,307 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x84\xd0\x00\x00\xbebj\x00\xc3\xadd\x86O\xa7\xdf$\x01\x00\x00\x00\xb2\xbe\x00\x00YH:@ _\xfa\x10;\xca+\xfb\xa0^5\x89')

03:42:11,325 __main__ DEBUG [Response]
(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe2\xd5\xfb\xff\xbeb\xee\x03\xc3\xadd\x86O\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\xca\x01\xc6\x1dd\xeb\xb7\xc6V\xda\xcb^\xd8D\xfa\x82', ('192.168.88.19', 80))

[... 41 lines cut ...]

03:42:18,492 __main__ DEBUG [Request]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00I\xd1\x00\x00\xbebj\x00\xcb\xadd\x86O\xa7\xdf$\x01\x00\x00\x00\xb3\xbe\x00\x00\xaf\xa8\xe9.DF\xe5\x19\xd5B\\i\x8ats\x04')

[... cut ...]
bytearray(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00K\xd1\x00\x00\xbebj\x00\xcd\xadd\x86O\xa7\xdf$\x01\x00\x00\x00\xb3\xbe\x00\x00\xaf\xa8\xe9.DF\xe5\x19\xd5B\\i\x8ats\x04')

03:42:20,562 __main__ DEBUG [Response]
(b'Z\xa5\xaaUZ\xa5\xaaU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xec\xd5\xfb\xff\xbeb\xee\x03\xcd\xadd\x86O\xa7\xdf$\x01\x00\x00\x00\xaf\xbe\x00\x00\xca\x01\xc6\x1dd\xeb\xb7\xc6V\xda\xcb^\xd8D\xfa\x82', ('192.168.88.19', 80))

03:42:21,563 __main__ DEBUG Failed to learn: no code received.
```

Can you help me to add the device type ?

Thank you.

felipediel, 2020-03-12:
@shuenpin I just updated the debug tool with your device type. Please [try again](https://drive.google.com/open?id=1Gr-Qe_sdVL-Yn2g2ITDFYB1TTpd1pX8Z).

shuenpin, 2020-03-12:
@faustinoaq 
it working fine!!!
BTW,the line 627 of the debug.py have the syntax error.

it should be 
```
        if devtype in [0x51da, 0x5f36, 0x610f, 0x62be]:
``

Thank you very much.

felipediel, 2020-03-12:
@shuenpin I just made some changes to improve the scalability of the code. Could you help me test it? Just download and run the debug again and check if everything is working. Ah, by the way, what is the name of your device? Ex: RM4c, RM4b.

shuenpin, 2020-03-12:
@felipediel When I executed new debug code,it show error message.
```
Traceback (most recent call last):
  File "debug3.py", line 1044, in <module>
    devices = discover(timeout=5)
  File "debug3.py", line 183, in discover
    dev = gendevice(devtype, host, mac)
  File "debug3.py", line 106, in gendevice
    return device_class(host=host, mac=mac, devtype=devtype)
  File "debug3.py", line 704, in __init__
    super().__init__(self, *args, **kwargs)
TypeError: __init__() got multiple values for argument 'host'
```
Should I use python2 or python3 ?
BTW,My device is RM4c.

felipediel, 2020-03-12:
Ah, sorry, I sent the wrong version. Again?

[56 more replies]

closed 2021-02-17 by felipediel