mjg59/python-broadlink #307: RM Mini 3 not discoverable or accepting auth if added using new Broadlink app

sebastian-silverwood, 2020-01-02:
RM Mini 3 must be added/configured initially using either this library, or the older IHC or e-Control apps. If you set up the RM Mini using the newer Broadlink app, it will discover and connect fine, but never authenticate. This prevents learning and sending of commands.

This is true for any projects that use this library too (for me, it was the broadlink component in Home Assistant, and where my digging started).

I'm not very technical, so not sure if this is an issue with this library that can be patched, or if it's something to do with the Broadlink app itself. Just wanted to share my experience if people have more experience debugging.

oTiMo, 2020-01-08:
thank you for this information. is it reversible?
I unfortunately used the newer application and I cannot use it

trankillity, 2020-01-08:
Yes, it's reversible. Hold down the reset button on the back then set up using either python-broadlink CLI or ihc or eControl.

oTiMo, 2020-01-08:
Thx, but it doesn't work. returned type is `0x5f36` it is really strange ...

````
Discovering...
###########################################
Unknown
# broadlink_cli --type 0x5f36 --host 192.168.1.21 --mac 5c4c7aa7df24

creemerica, 2020-01-09:
Same here, device type is 0x5f36. Added it to the list for RM3 devices however not able to learn any commands.

trankillity, 2020-01-09:
Strange! I'll admit that I set it up using the ihc app, which then allowed local auth/control (via Home Assistant's SmartIR addon, which uses python-broadlink).

oTiMo, 2020-01-09:
> Same here, device type is 0x5f36. Added it to the list for RM3 devices however not able to learn any commands.

exactly the  same thing for me

Nonobond007, 2020-01-09:
Hello I just bought an RM-Mini-3 on Amazon, broadlink_discovery returns me an rm device: 0x5f36.
I added this device to the init.py file, but it still doesn't work.

oTiMo, 2020-01-14:
see #308

dennisadvani, 2020-01-26:
I've received a few of the 0x5f36 type devices as well with firmware 44057. I can discover and configure them via python-broadlink but can't send any codes. Is anyone else able to send codes or is the issue you're experiencing only with learning them? Also, I initially configured via the new Broadlink app; after resetting and configuring via python, I still can't send codes.

creemerica, 2020-01-26:
> I've received a few of the 0x5f36 type devices as well with firmware 44057. I can discover and configure them via python-broadlink but can't send any codes. Is anyone else able to send codes or is [...]

Nope same thing here, I opened #308 regarding this same exact issue. Haven't been able to get it to learn new codes.

felipediel, 2020-04-19:
Fixed by #317.

felipediel, 2020-04-19:
This is a feature of the firmware of these new devices. Once connected to the cloud, they block local authentication. You need to [disable the cloud](https://github.com/home-assistant/core/issues/30215#issuecomment-581159499) in order to control them locally. There is nothing that can be done about it at library level.

felipediel, 2021-02-18:
When the device is locked - this is done automatically after authentication in the Broadlink app - it does not authenticate external applications, including this one. Now there is a button to unlock the device in the Broadlink app. This issue no longer exists. Closing.

closed 2021-02-18 by felipediel